dji-sdk/Onboard-SDK-ROS #6: Cant rosrun dji_sdk_manifold_read_cam

acdriec, 2016-01-19:
I removed the IGNORE file in dji_sdk_manifold_read_cam but when I do a catkin make, their is no dji_sdk_manifold_read_cam that i can rosrun.

lanyusea, 2016-01-19:
`catkin make clear`

then 

`catkin make` again and try

lanyusea, 2016-01-19:
and the node name is:

dji_sdk_read_cam

acdriec, 2016-01-20:
Thank you for you answer, It did a catkin_make clean and catkin_make but the node is still not created, when I type rosrun "tab" "tab" dji_sdk_read_cam is not shown

lanyusea, 2016-01-20:
1. did you try to run it instead of using `tab`?

I faced several times that there is no auto-complete but the pkg and node are actual there.
It works successfully after I type the whole package name. 

2. are other pkgs from this repo working well?

I know it may be a silly question but I'm still wanna asking if you forgot to source the `setup.bash` in your `bash.rc`

3. no, there is no launch file for this one.

acdriec, 2016-01-20:
Your silly question was the solution, sorry for that !

lanyusea, 2016-01-20:
glad it has been solved ;)

closed 2016-01-20 by lanyusea