dji-sdk/Onboard-SDK-ROS #8: cant subscribe on /dji_sdk/waypoint_navigation_action/feedback

acdriec, 2016-01-26:
When i try to subscribe on above topic I get following error:

[ INFO] [1453822400.085506030]: sdk_service_client_test
terminate called after throwing an instance of 'ros::ConflictingSubscriptionException'
  what():  Tried to subscribe to a topic with the same name but different md5sum as a topic that was already subscribed [dji_sdk/WaypointNavigationFeedback/28c39cf9bce64572a77ec405558cca9e vs. dji_sdk/WaypointNavigationActionFeedback/2bf531c7224b7f96b03a6df639e60db3]
Aborted

This is my code:
ros::Subscriber sub = nh.subscribe("/dji_sdk/waypoint_navigation_action/feedback", 1, waypointInfo);

and
void waypointInfo(const dji_sdk::WaypointNavigationFeedback::ConstPtr& msg)
{
  cout << "test " <<  msg->latitude_progress << endl;
}

I dont know if this is a ROS problem or something in the SDK or in my code?

groundmelon, 2016-01-26:
"WaypointNavigationFeedback"    vs    "WaypointNavigation**Action**Feedback"

"dji_sdk::WaypointNavigationFeedback"    vs    "/dji_sdk/waypoint_navigation_**action**/feedback"

acdriec, 2016-02-01:
Oke I see my fault, latitude progress can than be accessed with: msg->feedback.latitude_progress

closed 2016-02-08 by acdriec