dji-sdk/Onboard-SDK-ROS #9: GPS altitude not correct

acdriec, 2016-02-17:
I tried waypointnavigation. 
In simulation everything worked fine, also in real life everything went fine, except the GPS altitude () was not correct (-137m in Belgium ?!). Although the matrice went to the correct GPS coordinate and altitude (is guess it used baro altitude ?). But the waypoint altitude and altitude-progress (waypoint_navigation_feedback.altitude_progress) was not correct. Any idea what could be wrong ?

lanyusea, 2016-02-18:
You can select the way of calculating altitude in N1 Assistant. fused or raw.

Could you specify what the "be wrong" is.
If you mean it cannot reach 100% and keeps being 99% or whatever, you can increase the threshold in action.cpp file. It should solve it.

acdriec, 2016-02-18:
Well, latitude and longitude are correct, but altitude is not correct (in the message). It says that I am at - 137m and the waypoint progress of altitude stays at around 7%. Although the matrice goes to the correct altitude. 

Thank you for you reply!

lanyusea, 2016-02-18:
em... that's quite interesting.

Could you use `ros echo` to check the values of `height` and `altitude` in global_position topic, together with its `health flag`?

acdriec, 2016-02-19:
I will test this on monday,

Thank you

lanyusea, 2016-03-13:
@acdriec  how is everything now?

acdriec, 2016-03-16:
I will try it today (on an other location). Sorry for the delay !

lanyusea, 2016-03-16:
@acdriec  no rush. I'm just wondering if everything goes well or there is really some hiding bugs in our program. ;)

acdriec, 2016-03-16:
Healt was 4 today and height was correct. But we tested at an other location. I let you know if it happens again. Thanks!

closed 2016-04-13 by lanyusea