dji-sdk/Onboard-SDK-ROS #10: attitude_control assumes GPS

acdriec, 2016-03-04:
Hi, 

I noticed that when we want to use drone->attitude_control(0x40, x, x, x, x); (both horizontal and vertical velocity) that we need GPS. (Test: works outdoor with GPS but not indoor without GPS). Is there a possibility to use it without GPS ? Or another way to send pitch, roll speeds to the matrice ?

Kind Regards,
Dries

acdriec, 2016-03-07:
This is in branch 2.3, I see that you can use the virtual RC channels in 3.1 does this work without GPS. And is there a way to do it in branch 2.3 without GPS ?

lanyusea, 2016-03-07:
yes, you can control the roll/pitch angle directly without GPS.
if you want to use the velocity/position control mode, GPS signal is
necessary.

acdriec, 2016-03-07:
You mean in HORI_ATTI_TILT_ANG mode ?

lanyusea, 2016-03-07:
yes

acdriec, 2016-03-07:
Ok we will test this very soon, thanks !

lanyusea, 2016-03-07:
but BE CAREFUL!

the drone cannot stable itself when in this mode.

acdriec, 2016-03-07:
Works like a charm !

closed 2016-03-07 by acdriec

jasperza, 2016-03-09:
If we are indoors with no GPS but have guidance in opti mode, what method should we use to control the drone?

lanyusea, 2016-03-10:
@jasperza I recommend the velocity control