dji-sdk/Onboard-SDK-ROS #11: Seg fault when run dji_sdk_node

buqing2009, 2016-03-09:
`
NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [18812]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e49b2e2e-e5ab-11e5-a172-448a5b46c753
process[rosout-1]: started with pid [18825]
started core service [/rosout]
process[dji_sdk-2]: started with pid [18842]
[dji_sdk-2] process has died [pid 18842, exit code -11, cmd /home/buqing2009/dji_ws/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/buqing2009/.ros/log/e49b2e2e-e5ab-11e5-a172-448a5b46c753/dji_sdk-2.log].
log file: /home/buqing2009/.ros/log/e49b2e2e-e5ab-11e5-a172-448a5b46c753/dji_sdk-2*.log
`

lanyusea, 2016-03-09:
make the enable_groundstation in launch file be 0.
and turn off the groundstation msgs in the SDK page of DJI Assistant2.

we have heard one crash case of -11 on ROS before, but cannot reproduce it.

groundmelon, 2016-03-09:
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7bcc02c in DJI::onboardSDK::CoreAPI::setWayPointCallback (this=0x3c3e6d617261703c, 
    handler=0x70d346 <DJI::onboardSDK::ROSAdapter::missionStatusCallback(DJI::onboardSDK::CoreAPI*, DJI::onboardSDK::Header*, void*)>, userData=0x9f9980)
    at /home/ltb/catkin_ws/src/dji_sdk/src/sdk_lib/DJI_App.cpp:167
167	    wayPointCallback.callback = handler;
(gdb) bt
#0  0x00007ffff7bcc02c in DJI::onboardSDK::CoreAPI::setWayPointCallback (this=0x3c3e6d617261703c, 
    handler=0x70d346 <DJI::onboardSDK::ROSAdapter::missionStatusCallback(DJI::onboardSDK::CoreAPI*, DJI::onboardSDK::Header*, void*)>, userData=0x9f9980)
    at /home/ltb/catkin_ws/src/dji_sdk/src/sdk_lib/DJI_App.cpp:167
#1  0x0000000000710e2e in DJI::onboardSDK::ROSAdapter::setMissionStatusCallback<DJISDKMission> (this=0x9f9980, func=
    (void (DJISDKMission::*)(DJISDKMission * const, uint8_t *, uint8_t)) 0x70bf34 <DJISDKMission::mission_status_callback(unsigned char*, unsigned char)>, 
    obj=0xa1cf80) at /home/ltb/catkin_ws/src/dji_sdk/include/dji_sdk/DJI_LIB_ROS_Adapter.h:126
#2  0x000000000070d0f4 in DJISDKMission::DJISDKMission (this=0xa1cf80, nh=...) at /home/ltb/catkin_ws/src/dji_sdk/src/modules/dji_sdk_node_mission.cpp:343
#3  0x0000000000661480 in DJISDKNode::DJISDKNode (this=0x9f9a50, nh=..., nh_private=...) at /home/ltb/catkin_ws/src/dji_sdk/src/modules/dji_sdk_node_main.cpp:278
#4  0x0000000000651486 in main (argc=1, argv=0x7fffffffd8e8) at /home/ltb/catkin_ws/src/dji_sdk/src/dji_sdk_node.cpp:17

Same problem (exit code -11) on my laptop with enable_groundstation=1. Hope this trace is helpful.

buqing2009, 2016-03-09:
OK, it seems that the problem is solved. Thank you! When the bug is fixed, i will close this issue.

groundmelon, 2016-03-09:
in this line: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.1/dji_sdk/src/modules/dji_sdk_node_main.cpp#L276

It will try to call rosAdapter->coreAPI->setWaypointCallback(...), but rosAdapter->coreAPI will not be constructed until https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.1/dji_sdk/src/modules/dji_sdk_node_main.cpp#L279

Another problem is, in the same line #276,  ```DJISDKMission* dji_sdk_mission``` will be destroyed when you go out of the ```if(...){...}```, and we cannot point to it again. This might cause memory leak and some potential bugs in the future. Maybe a class member shared_ptr is better?

Pointers are really dangerous......

lanyusea, 2016-03-09:
then you must be a genius of pointer expert.

XD

buqing2009, 2016-03-09:
I guess you guys are buddies(基友)...

lanyusea, 2016-03-09:
#12 should be fixed now. @buqing2009

buqing2009, 2016-03-09:
@lanyusea Yeah, it's ok when run the node. I have not tested the function about ground station, but it's not in category of this issue, so i will close it, thank you guys @lanyusea @groundmelon . If any problem i met later, i will reopen it.

closed 2016-03-09 by buqing2009