dji-sdk/Onboard-SDK-ROS #13: activate SDK version did not match

acdriec, 2016-04-01:
Hi,

I have installed the DJI SDK ROS on a brix minicomputer. And I try to run "roslaunch dji_sdk sdk_manifold.launch"

But I get the next error and cant find whats wrong : 

=================================================
--- Connection Info ---
Serial port: /dev/ttyUSB0
Baudrate: 230400
-----
STATUS init,line 41: going to open device /dev/ttyUSB0 with baudrate 230400...
STATUS init,line 46: ...succeed to start serial
[ INFO] [1459513193.703278180]: Succeed to create thread for readPoll
STATUS getVersionCallback,line 222: version ack = 65281
STATUS getVersionCallback,line 223: version crc = 0xD3D21804
STATUS getVersionCallback,line 225: version ID = SDK-v1.0 BETA M100-02.03.10.00
STATUS getVersionCallback,line 227: version name = TA M100-02.03.10.00
**ERROR activateCallback,line 276: activate SDK version did not match**
STATUS getVersionCallback,line 222: version ack = 65281
STATUS getVersionCallback,line 223: version crc = 0xD3D21804
STATUS getVersionCallback,line 225: version ID = SDK-v1.0 BETA M100-02.03.10.00
STATUS getVersionCallback,line 227: version name = TA M100-02.03.10.00

Thansk for your help

lanyusea, 2016-04-01:
please update your firmware of M100

acdriec, 2016-04-01:
Ah I see, I first had to install the latest version of the DJI Assistant, than i could see that my firmware was not the latest.

thanks !

closed 2016-04-05 by lanyusea

janwillemm, 2016-05-02:
We have the same problem, and we have installed the latest firmware on the Matrice 100 using the Assistant 2 software.

"ERROR activateCallback,line 302: activate SDK version did not match
STATUS getDroneVersionCallback,line 252: version ack = 65281
STATUS getDroneVersionCallback,line 253: version crc = 0xA6453AAC
STATUS getDroneVersionCallback,line 255: version ID = 041D491989
STATUS getDroneVersionCallback,line 257: version name = SDK-v1.0 BETA M100-03.01.01.00"

We are now downgrading the software to see if that helps..

lanyusea, 2016-05-02:
@janwilemm please use the latest sample code with the latest firmware downloaded from the latest DJI Assistant

janwillemm, 2016-05-03:
We've tried with the downgrade, which didn't work. We've updated the Assistent, the firmware and used the latest Onboard SDK from github and now we aren't able to communicate any more..

"Error encrypt, line 851: Can not send encode data, no available key." I'm using the commandLine example and the commands i do are:

--SS load key.txt (which confirms the loading of the correct key)
--CA ac (No response)
--CA ct 1 (Gives the error)

What can we do?

lanyusea, 2016-05-03:
@janwillemm enable API in your DJI Assistant

janwillemm, 2016-05-03:
Well, indeed, after reinstalling the firmware, we forgot to enable the API in the DJI assistant. It works now, thanks!

lanyusea, 2016-05-03:
@janwillemm  you're welcome :)

jeyong, 2016-06-07:
@janwillemm In my case ROS example works but Linux commandline example doesn't work. I got same problem as you. Does it work well in Linux commadline example?

janwillemm, 2016-06-13:
@jeyong Yeah, I've used the Linux commandline example and it works ok. But it's a time ago, so not sure what steps I took to make it working...