dji-sdk/Onboard-SDK-ROS #15: Cant take off after arming

acdriec, 2016-04-06:
Is it normal that you cant take off after an arming command ? 
drone->drone_arm();
drone->takeoff();

It is not really a problem because you can take off without arming but isn't it strange ?

Kind Regards

lanyusea, 2016-04-06:
you can take off directly without arming the drone.

because they are two different APIs.

as for it cannot take off after arming, em.. I'm not sure and will test it later.
If so, I think it should be a feature in flight controller.

lanyusea, 2016-04-06:
@acdriec  yes you're right.
I confirmed with DJI that it is a feature of takeoff function, it only works when motor isn't running.

also, the disarm command only works when drone is not in the air, I updated the document to make it more clear: https://github.com/dji-sdk/Onboard-SDK/commit/1214689468eec9e456fce738bcee756db4b0fc75

closed 2016-04-13 by lanyusea