dji-sdk/Onboard-SDK-ROS #16: gimbal_angle_control

acdriec, 2016-04-15:
I try to turn the gimbal downwards by sending:
gimbal_angle_control(0,-900,0,10); 
This works but when I rotate the Matrice (yaw axis), the camera keeps its position towards the ground by rotating its yaw axis. I dont want this, I want the camera to rotate with the matrice, so yaw axis fixed wrt the Matrice, so I tried this:
gimbal_angle_control(0,-900,0,10,0,1);
But this is also not working, any idea what I have to try ?

Kind Regards

lanyusea, 2016-04-15:
as I remember, the gimbal API doesn't provide a selection of Ground Frame of Body Frame.

could you try to adjust the gimbal mode in DJI GO?
maybe using the "follow mode" may work, but I'm not sure

acdriec, 2016-04-15:
No it does not work, i seem like for the Matrice there is no difference between follow, fpv and free, gimbal is always trying to keep level.

I mentioned that it is the yaw of the gimbal that i try to fix but when the cam is facing forward it is the pan ofcourse.

groundmelon, 2016-04-15:
I suggest that you can try out the gimbal speed control interface. You can write a feedback controller to lock the yaw angle w.r.t body, and just give gimbal speed command to the drone. However, the yaw angle data sent from the drone is not very precise in my experience.

acdriec, 2016-04-16:
And can we find/read the angle of the gimbal w.r.t. the drone somewhere ? Than it is also possible to write a feedback controller.

groundmelon, 2016-04-16:
There is no reading of angle of the gimbal w.r.t body, but there is reading of angle of the gimbal w.r.t world. You can use rotation minus to get it. But this does not have high precision.

closed 2016-04-27 by lanyusea