dji-sdk/Onboard-SDK-ROS #19: Compass reading is not published

Asiron, 2016-04-25:
We don't receive any data on the topic `/dji_sdk/compass`. However we do get GPS data on topic `dji_sdk/global_position`.

With the following system configuration:

M100 and Manifold
Ubuntu 14.04 / ROS Indigo
Onboard-SDK-ROS [https://github.com/dji-sdk/Onboard-SDK-ROS/commit/43ac30cc5ab74b3aebdadf854a71e12fb61dc69f](url)

Flight Controller Firmware version `1.3.1.00`, updated using DJI Assistant Software v2.0

We registered the DJI App through DJI Developer Account and it successfully activates.
What seems strange is that we get this output when we check the version with 
`rosservice call /dji_sdk/version_check`

```
STATUS getSDKVersionCallback,line 249: version ack = 0STATUS getSDKVersionCallback,line 250: version crc =│
 0xA6453AACSTATUS getSDKVersionCallback,line 252: version ID = 041DD22937STATUS getSDKVersionCallback,line│
 255: version name = SDK-v1.0 BETA M100-03.01.01.00 
```

It shows SDK version `SDK-v1.0`, isn't it supposed to show the firmware version `1.3.1.00`?

lanyusea, 2016-04-25:
[image]

read document first, please

Asiron, 2016-04-25:
We set this to 50Hz already as it's mentioned in the manual.

lanyusea, 2016-04-25:
@Asiron check if RC data and battery data also correct.
if so, you didn't set compass to 50Hz successfully.

Asiron, 2016-04-25:
Please find below our setting. We replicated all except RTK, which is replicating the GPS setting.
Also `Battery Level` and `GPS` is set to not send, but we receive messages on its corresponding topics.
[http://i.imgur.com/0OzNeCC.jpg](url)

lanyusea, 2016-04-25:
is the value you receive correct or not?

Asiron, 2016-04-25:
We are getting correct battery level. As for the postion, we are getting the following message
```
header: 
  seq: 4758
  stamp: 
    secs: 1461599984
    nsecs: 902397126
  frame_id: /world
ts: 360936
latitude: 0.0
longitude: 0.0
altitude: 396.423736572
height: 0.0
health: 0
---
```
However, we are indoor, since it's raining outside.

lanyusea, 2016-04-25:
the battery is in the last position of broadcast data struct.
if the battery is correct then there won't be a struct-shift problem and the code works well.

then it should be something related to your DJI Assistant which didn't set the frequency correctly. no idea why it happen.

btw, you can check the latitude/longitude correct or not by the simulator in DJI Assistant

Asiron, 2016-04-25:
We checked and we are getting correct lat/long with the simulation.

lanyusea, 2016-04-26:
@Asiron  not your fault.
I will kick the ball into DJI Assistant team

lanyusea, 2016-04-26:
@Asiron 
should be fixed now.
The Assistant will upgrade by itself (may need a reboot).

Once you find the GPS/RTK is gone, then everything will be fine.
The GPS information is published out from the `Position`.

[image]

Asiron, 2016-04-26:
Unfortunately, we are unable to upgrade The Assistant. We rebooted and disabled firewall with no effect.
We tried on different computers as well.

Would be it possible to get a binary of the new version?

lanyusea, 2016-04-26:
redownload from the official website again may work

Asiron, 2016-04-26:
Sorry I forgot to mention, we did that and it didn't work anyway.

lanyusea, 2016-04-26:
@Asiron then you can use the old one and wait for an upgrade that will happen some time in the future.

Just forget the `GPS` and `RTK` and lift everything below them up two places.
i.e. GPS stands for mag, RTK stands for RC and so on.

Asiron, 2016-04-26:
anyway, if we set the magnetometer (GPS) to 50Hz or 10Hz we don't get anything in the compass rostopic

phil333, 2016-04-27:
I have the same problem, is there any update on this?

lanyusea, 2016-04-27:
@phil333 The DJI Assistant has already been upgraded to fix this problem.

phil333, 2016-04-27:
Well, we still have DJI Assistant 2 v.1.0.3 from 2016-04-07 from the website. We have not been able to update the software itself. Rebooting doesn't update the software and neither does running the "DJISoftwareUpdater.exe". A separate installer or an ETA on the update would be helpful.

lanyusea, 2016-04-28:
should be okay now.

sorry for the delay, they are waiting for an support related to M600 in this new version

Asiron, 2016-04-29:
@lanyusea Thanks a lot, the compass is working now

closed 2016-04-29 by lanyusea

phil333, 2016-04-29:
working here too, thx for the quick response