dji-sdk/Onboard-SDK-ROS #21: ctakin_make errors

sytelus, 2016-04-26:
I'm getting below errors when running catkin_make on latest version:

```
[100%] Building CXX object Onboard-SDK-ROS/dji_sdk_web_groundstation/CMakeFiles/dji_sdk_web_client.dir/src/map_nav_srv.cpp.o
Linking CXX executable /home/shitals/vso/msresearch/Theseus/catkin_ws/devel/lib/run_depth_2_cloud/run_depth_2_cloud_node
[100%] Built target run_depth_2_cloud_node
Linking CXX executable /home/shitals/vso/msresearch/Theseus/catkin_ws/devel/lib/dji_sdk_dji2mav/minimal
[100%] Built target minimal
In file included from /opt/ros/indigo/include/ros/ros.h:40:0,
                 from /home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_dji2mav/src/dji2mav_bringup.cpp:15:
/home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_dji2mav/src/dji2mav_bringup.cpp: In function ‘void respondToWpTarget(const float (*)[7], uint16_t, uint16_t)’:
/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%d’ expects argument of type ‘int’, but argument 8 has type ‘std::vector<dji_sdk::MissionWaypoint_<std::allocator<void> >, std::allocator<dji_sdk::MissionWaypoint_<std::allocator<void> > > >::size_type {aka long unsigned int}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/indigo/include/ros/console.h:345:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(0, __VA_ARGS__)
     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:557:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_dji2mav/src/dji2mav_bringup.cpp:117:5: note: in expansion of macro ‘ROS_INFO’
     ROS_INFO("Size of the wpl: %d", task.mission_waypoint.size());
     ^
In file included from /opt/ros/indigo/include/ros/ros.h:40:0,
                 from /home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_web_groundstation/src/map_nav_srv.cpp:1:
/home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_web_groundstation/src/map_nav_srv.cpp: In function ‘void goalCB()’:
/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 8 has type ‘uint64_t {aka long unsigned int}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/indigo/include/ros/console.h:345:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(0, __VA_ARGS__)
     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:557:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_web_groundstation/src/map_nav_srv.cpp:81:13: note: in expansion of macro ‘ROS_INFO’
             ROS_INFO("Cancel task with tid %llu", tid_);
             ^
/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 8 has type ‘uint64_t {aka long unsigned int}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/indigo/include/ros/console.h:345:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(0, __VA_ARGS__)
     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:557:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_web_groundstation/src/map_nav_srv.cpp:87:17: note: in expansion of macro ‘ROS_INFO’
                 ROS_INFO("Start task with tid %llu", tid_);
                 ^
/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 8 has type ‘uint64_t {aka long unsigned int}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
[... 50 lines cut ...]
                 ROS_INFO("Set task(if any) with tid %llu preemted", newGoal.tid);
                 ^
/home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_web_groundstation/src/map_nav_srv.cpp: In function ‘void cmdCB(const MapNavSrvCmdConstPtr&)’:
/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 9 has type ‘dji_sdk_web_groundstation::MapNavSrvCmd_<std::allocator<void> >::_tid_type {aka long unsigned int}’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/indigo/include/ros/console.h:345:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(0, __VA_ARGS__)
     ^
/opt/ros/indigo/include/ros/console.h:375:7: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION’
       ROSCONSOLE_PRINT_AT_LOCATION(__VA_ARGS__); \
       ^
/opt/ros/indigo/include/ros/console.h:557:35: note: in expansion of macro ‘ROS_LOG_COND’
 #define ROS_LOG(level, name, ...) ROS_LOG_COND(true, level, name, __VA_ARGS__)
                                   ^
/opt/ros/indigo/include/rosconsole/macros_generated.h:110:23: note: in expansion of macro ‘ROS_LOG’
 #define ROS_INFO(...) ROS_LOG(::ros::console::levels::Info, ROSCONSOLE_DEFAULT_NAME, __VA_ARGS__)
                       ^
/home/shitals/vso/msresearch/Theseus/catkin_ws/src/Onboard-SDK-ROS/dji_sdk_web_groundstation/src/map_nav_srv.cpp:201:5: note: in expansion of macro ‘ROS_INFO’
     ROS_INFO("Received command \"%c\" of tid %llu", msg->cmdCode, msg->tid);
     ^
Linking CXX executable /home/shitals/vso/msresearch/Theseus/catkin_ws/devel/lib/dji_sdk_dji2mav/dji2mav_bringup
[100%] Built target dji2mav_bringup

```

lanyusea, 2016-04-27:
excuse me, where is the error?
it says compilation is done and start linking.

closed 2016-05-04 by lanyusea

sytelus, 2016-05-11:
It's warnings. Some of us like to turn on warnings as errors just so they don't ignored. I think DJI SDK team should definitely consider that :).

`/opt/ros/indigo/include/ros/console.h:342:176: warning: format ‘%d’ expects argument of type ‘int’, but argument 8 has type ‘std::vector<dji_sdk::MissionWaypoint_<std::allocator<void> >, std::allocator<dji_sdk::MissionWaypoint_<std::allocator<void> > > >::size_type {aka long unsigned int}’ [-Wformat=]`