dji-sdk/Onboard-SDK-ROS #23: simulator not working with the manifold

Annaiyan, 2016-04-27:
Hi,
I have installed onboard sdk ros in the manifold and connected the M100 to the dji pc simulator using micro usb cable.i launched the dji_sdk_client (in the demo) and tried with various options like take off,arm/disarm, the simulator is not reacting.Besides,i have connected the remote controller with function mode with the dji go app.For your information,the simulator is working with rc and dji go app except with manifold.
How to activate the manifold controls in the dji pc simulator.
Is there any solution for this problem?

lanyusea, 2016-04-27:
run the dji_sdk node before dji_sdk_demo

Annaiyan, 2016-04-28:
thanks!it is working fine now.

closed 2016-04-28 by Annaiyan