dji-sdk/Onboard-SDK-ROS #25: Control mode clarification

AlexandreBorowczyk, 2016-05-19:
Hi,

I would like to clarify the control modes.

the onboard-sdk Appendix mentions commands can be given either in the ground frame or the body frame. I assume the ground frame to the NED such as the x, y input will result is motion along the NED frame. However, I'm confused about control behavior in the body frame mode. During testing it seem that the yaw command impact the direction of motion. Also what is the impact of non-stable mode vs stable mode?

Thanks

lanyusea, 2016-05-19:
1. it's NEU not NED
2. the yaw axis has no 'body frame'
3. it will make your control behavior more smooth

AlexandreBorowczyk, 2016-05-19:
2. I understand for the yaw but it seems that in HOR_POS the input [1x, 90 yaw] wont give the same result as [1x, 0 yaw] even is the M100 start with the same orientation. for the first it would move "right" and the second would move front. Futhermore, is de body axis rotation with the matrice 100 such as is the M100 is at a -30 deg pitch angle and we give a 1x input, would it climb?

Thanks

lanyusea, 2016-05-19:
1. if you're using body frame, possible.
2. yes if 1x here means increasing throttle

AlexandreBorowczyk, 2016-05-19:
Sorry I wasn't very clear. By 1x, I meant a cmd of 1 meter along X-axis

lanyusea, 2016-05-19:
well, you cannot control angle or velocity in the same time. it must be either HORI_ATTI or HORI_VEL ( or POSI).

what's more, x and roll are in the same parameter position, y and pitch are the same.

therfore if you want an attitude angle control of pitch, you cannot achieve a velocity control in y axis to climb up at the same time.

AlexandreBorowczyk, 2016-05-19:
I see, thanks it's much more clear for me!

closed 2016-05-19 by AlexandreBorowczyk