dji-sdk/Onboard-SDK-ROS #29: [Problem] DJI SDK-ROS Mavlink example

won13y, 2016-06-09:
I was able to successfully run the dji-sdk-demo in Onboard-SDK-ROS 

But, I don't know how to run the following examples:

https://github.com/dji-sdk/Onboard-SDK-ROS/tree/3.1/dji_sdk_dji2mav

I was able to successfully run the following commands:

    roslaunch dji_sdk sdk_manifold.launch

But the problem occurs in following webpages: 

https://github.com/dji-sdk/Onboard-SDK-ROS/tree/3.1/dji_sdk_dji2mav

it indicates as followings:

    Record the IP and port of ground control station, modified the corresponding param in dji2mav_bringup.launch. Then run:

    roslaunch dji_sdk_dji2mav dji2mav_bringup.launch

Especially, I don't know the what the IP and port of ground control station means in above sentences
(How can I find the the IP and port of ground control station?)
 
I was able to find the IP by ifconfig command, but was not able to find the port.

Does the IP and port of ground control station means those of the PC which is running Qgroundcontrol? 

Actually, I changed the IP(which I found by "ifconfig" command) and port of the PC which is running Qgroundcontrol in dji2mav_bringup.launch file, but it failed.

    <arg name="targetIp1" default="10.60.23.136" /> 
    <arg name="targetPort1" default="14550" /> 
    <arg name="srcPort1" default="14551" />

pcler, 2016-06-11:
> Does the IP and port of ground control station means those of the PC which is running Qgroundcontrol?

Yes. The target IP/port mean the ones of your PC that is running Qgroundcontrol Station. And the src port means the port of Onboard linux that you would like to communicate with the Qgroundcontrol Station. In the concrete, you can get target IP by checking the IP of your PC which is running Ogroundcontrol Station. And you can set target port by modifying the launch file that you mention, meanwhile input the same value on your Qgroundcontrol Station as the 4th point of https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.1/dji_sdk_dji2mav/README.md shows.

What's more, it is allowed that several stations connect to the same drone. That is how the indexes after "targetIp" "targetPort" "srcPort" come.

If the connection is fail, please double check whether the IP and port is correctly config. Then check out whether anything else blocks the port. Some firewalls may do so.

pcler, 2016-06-11:
It would be better if you could show us more details about the failure, such as your launch file and your Qgroundcontrol station config. Thank you!

lanyusea, 2016-07-03:
closed because we have updated all codes/docs about the onboard SDK to solve any possible problem.

please reopen the issue if you still suffer the same problem

closed 2016-07-03 by lanyusea

won13y, 2016-09-07:
[image]