dji-sdk/Onboard-SDK-ROS #31: Issues with 0.1.8 Release

uwleahcim, 2016-06-14:
I was testing out the 0.1.8 release and noticed that dji_sdk topics are completely broken. Is there firmware update for the M100 that is suppose to coincide with the 0.1.8 release, am I missing anything, or 0.1.8 has bugs?

Thank you

lanyusea, 2016-06-14:
bug.

kicking the ass of @bobbyshashin , repairing.

lanyusea, 2016-06-14:
@uwleahcim 

problem is: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.1/dji_sdk_lib/src/DJI_App.cpp#L105 

    passData(*enableFlag, dataflag, &broadcastData.gimbal, pdata,
             sizeof(GimbalData) - (versionData.version == versionM100_23) ? 1 : 0, len);

should be 

    passData(*enableFlag, dataflag, &broadcastData.gimbal, pdata,
             sizeof(GimbalData) - ((versionData.version == versionM100_23) ? 1 : 0), len);

together with the battery following.

You can update it by yourself or wait for an upgrade from us.

bobbyshashin, 2016-06-14:
Bug fixed. Thanks for the notification @uwleahcim

closed 2016-06-21 by uwleahcim