dji-sdk/Onboard-SDK-ROS #34: Attitude Control HORI_POS

skyritsis, 2016-06-30:
Hello,
In the examples, the attitude control is within a loop. But from what I can understand if someone wants to work with HORI_POS flag, the developer should just send one command with the offset in meters and wait for it to be finished. Though, when simulating this, it does not seem to work as intended.
Is there something I am missing?
Thank you in advance,
-SK

lanyusea, 2016-06-30:
your understanding is incorrect

skyritsis, 2016-06-30:
That, I can see. Though you did not seem to clarify what I am missing.

lanyusea, 2016-06-30:
it should be in a loop, as mentioned in the document

lanyusea, 2016-06-30:
every control command will execute the drone for 0.02 second, no matter position, velocity or attitude. 

therefore, you need to update the position offset and send the updated offset to the drone in 50 Hz

skyritsis, 2016-06-30:
Thank you very much. This was brilliant.
Good day.

closed 2016-06-30 by skyritsis