dji-sdk/Onboard-SDK-ROS #39: Gimbal publishing wrong yaw?

uwleahcim, 2016-07-13:
I've been having huge issues with my project. 

Apparently there is a bug that causes the gimbal at 0,0,0 to publish 0, -156.6, 0 (pitch, yaw, and roll).

I've tested this out with a fresh download of Onboard-SDK-ROS

In the simulator, the gimbal reports correctly 0,0,0 starting position. But in real world, the gimbal report 0, -156.6, 0 at starting position. I've tried calibrating the gimbal in the DJI Go App, but nothing seems to fix the issue.

lanyusea, 2016-07-14:
well, the reason is that 
1. the yaw published from the drone is not in drone coordinate, but in NED coordinate
2. the simulator cannot reach the gimbal, gimbal has its own IMU and it can't be simulated.

uwleahcim, 2016-07-14:
Sorry, I'm still confused on exactly what reference the gimbal's yaw is in. Is it suppose to be ground NED or body NED (drone) (i.e. is yaw suppose to be 0 when the gimbal is pointed North, or when gimbal is pointed forward on the drone)? I guess I've always assumed it was in body coordinates as services calls like gimbal_angle_control are in body coordinates i.e. yaw = 0 means camera points forward (I just tested roll and pitch in gimbal_angle_control and they're in ground coordinates, but yaw is in body coordinates...). 

Anyways, it looks like all rotational axis are in ground coordinates (roll, pitch and yaw), but for gimbal_angle_control yaw is in body coordinates. 

Anyways, if the gimbal has its own IMU generating data in the ground frame, I need to rebuild define my tf of the gimbal-->base to gimbal-->odom (the vector3 of the translation will be based on local position of the body plus offset of the gimbal). This probably explains why in my test flights the PID controls were way off. I guess I have two options here. Re-define gimbal frame in the tf to be linked to the odom/map frame. Or subtract out the attititude_quaternion from the gimbal's rotation to force to gimbal frame to be linked to the body frame.

lanyusea, 2016-07-14:
well, when I say NED I mean the grouind frame (yaw is 0 when facing North) instead of body frame (yaw is 0 when facing drone's head)

and the gimbal_angble_control also uses NED frame.

uwleahcim, 2016-07-14:
Well I just want to report back that modified my tf tree to link gimbal to the odom frame (which is also my map frame) and now things are performing as expected like in the simulator (except now my simulator is broken, so I have to ignore the gimbal frame). 

As for gimbal_angle_control, it is also NED except for yaw it seems, i.e. gimbal_angle_control(###, ###, 0, true), which is absolute mode, will always make the gimbal point forward with respect to the M100.

closed 2016-07-14 by uwleahcim