dji-sdk/Onboard-SDK-ROS #40: error during initilaization

Aswath93, 2016-07-18:
Hey,

I am at the activation step that needs to be done before using onboard-sdk-ros. I have modified the launch file "dji_sdk/launch/sdk_manifold.launch" accordingly.Now, When I try "roslaunch dji_sdk sdk_manifold.launch" I get the following error `ERROR _serialOpen,line 129: cannot open device /dev/ttyTHS1
ERROR init,line 44: ...fail to start serial
`
could anyone tell me what to do now?

Thanks

lanyusea, 2016-07-21:
you serial port is not configured successfully.

It start read the port, but nothing received, then stuck here.

Aswath93, 2016-07-22:
thanks for the reply,could you tell me how to configure it correctly?

lanyusea, 2016-07-22:
connect your mobile with the RC, open DJI GO in your mobile.

lanyusea, 2016-07-22:
r u using Manifold?

Aswath93, 2016-07-23:
I did have my tablet connected to the paired rc with the dji go app open.
Anything else that I might be missing?

Aswath93, 2016-07-23:
no i am trying to connect to my linux pc.

lanyusea, 2016-07-23:
so, if you choose the correct serial port device, and open 'enable API' in DJI Assistant2

it may be your USB-TTL module doesn't work

Aswath93, 2016-07-26:
I read on the dji developer forum that the signal voltage might be a problem. I am using the USB-TTL cable recommended by dji in the hardware guide. Is there another cable that you recommend?

amenon7, 2016-07-26:
@Aswath93 : 

1.) Ensure serial device ID is the same as your launch file. 
2.) Ensure Enable API is set and you have the correct baud rate set. 
2.) Ensure RC is paired and DJI GO is running and connected to the internet. 
3.) Ensure RC is in F mode. OnboardSDK only works only in F mode. 

If you are using the recommended USB-TTL cable, you should not have a problem with your cable.

Aswath93, 2016-07-27:
Hey Arjun,

Thank you for replying.

I did all those things.

I am finding the serial ID using "lsusb" command, and setting the permissions using "chmod" command.

I do not know what else to do!

Is there anything else I might have missed?

amenonDJI, 2016-08-22:
@Aswath93 Have you tried this: To access the serial port, add your username to the dialout group by typing sudo usermod -a -G dialout $USER (you do not need to replace $USER with your username). Then logout and login again.

closed 2016-12-01 by amenonDJI