dji-sdk/Onboard-SDK-ROS #41: [Onboard Control Request Problem] STATUS activateCallback,line 280: new device, please link DJIGO to your remote controller and try again

won13y, 2016-07-21:
I am using DJI Matrice 100,

and Onboard computer is intel NUC,

When I run dji_sdk_demo, and following occurs

In particular, following warning is produced

    STATUS activateCallback,line 280: new device, please link DJIGO to your remote controller and try again

Thus, I cannot request control from RC to onboard computer

Would anyone of you help me out?

The entire output of command is as followings:

wkyoun@wkyoun:~/catkin_ws/src$ roslaunch dji_sdk sdk_dmm.launch

    ... logging to /home/wkyoun/.ros/log/2616ca64-4f03-11e6-9bd3-b8aeedea2130/roslaunch-wkyoun-5105.log
    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.

    started roslaunch server http://192.168.43.252:33820/

    SUMMARY
    ========

    PARAMETERS
     * /dji_sdk/app_bundle_id: Welcome to use dj...
     * /dji_sdk/app_id: 1030093
     * /dji_sdk/app_version: 1
     * /dji_sdk/baud_rate: 230400
     * /dji_sdk/enc_key: cf02c43d625667ecb...
     * /dji_sdk/groundstation_enable: 1
     * /dji_sdk/serial_name: /dev/ttyUSB0
     * /rosdistro: indigo
     * /rosversion: 1.11.19

    NODES
      /
        dji_sdk (dji_sdk/dji_sdk_node)
        dji_sdk_client (dji_sdk_demo/dji_sdk_client)
        dji_tf (dji_tf/dji_tf)
        rviz (rviz/rviz)
    
    auto-starting new master
    process[master]: started with pid [5117]
    ROS_MASTER_URI=http://192.168.43.252:11311

    setting /run_id to 2616ca64-4f03-11e6-9bd3-b8aeedea2130
    process[rosout-1]: started with pid [5130]
    started core service [/rosout]
    process[dji_sdk-2]: started with pid [5141]
    process[dji_sdk_client-3]: started with pid [5148]
    [ INFO] [1469078562.919109662]: sdk_service_client_test
    process[dji_tf-4]: started with pid [5156]
    process[rviz-5]: started with pid [5167]
    =================================================
    app id: 1030093
    app version: 0x03010A00
    app key: cf02c43d625667ecb0525276c0dbf1e9ec83a63a4c5fc2cf36f5c4ffd1504511
    =================================================
    --- Connection Info ---
    Serial port: /dev/ttyUSB0
    Baudrate: 230400
-----
    STATUS init,line 41: going to open device /dev/ttyUSB0 with baudrate 230400...
    STATUS init,line 46: ...succeed to start serial
    [ INFO] [1469078563.119421840]: Succeed to create thread for readPoll
    STATUS getSDKVersionCallback,line 249: version ack = 65281STATUS getSDKVersionCallback,line     250: version crc = 0xA6453AACSTATUS getSDKVersionCallback,line     252: version ID = 041DD20869STATUS getSDKVersionCallback,line 255: version name = SDK-v1.0     BETA M100-03.01.01.00
    STATUS activateCallback,line 280: new device, please link DJIGO to your remote controller and try again

lanyusea, 2016-07-21:
please link DJIGO to your remote controller and try again

won13y, 2016-07-21:
sudo ifconfig eth0 192.168.3.100
sudo route add 192.168.1.201 eth0

amenon7, 2016-07-26:
@won13y Have you tried what @lanyusea recommended? If it works now, I'd like to close this issue.

amenonDJI, 2016-08-22:
Please re-open with description if the problem persists.

closed 2016-08-22 by amenonDJI

smithan7, 2017-05-07:
won13y's fix worked for me