dji-sdk/Onboard-SDK-ROS #42: Error during "catkin_make"

ItaloAW, 2016-07-27:
Hello everyone,

Im trying to run the cam package, but I'm facing a weird error. I've followed this tutorial [TUTO](https://developer.dji.com/onboard-sdk/documentation/github-platform-docs/ROS_Example/ros_video_decoding_package.html) but when I do "catkin_make" the following error appears:

OBS: yes, I did "source devel/setup.bash"
OBS1: "catkin_make clean" too
OBS2: I've checked all the dependences and they are all there

root@tegra-ubuntu:/home/ubuntu/workspace# catkin_make
Base path: /home/ubuntu/workspace
Source space: /home/ubuntu/workspace/src
Build space: /home/ubuntu/workspace/build
Devel space: /home/ubuntu/workspace/devel
Install space: /home/ubuntu/workspace/install
WARNING: Package name "FCS" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/workspace/build"
####
####
#### Running command: "make -j2 -l2" in "/home/ubuntu/workspace/build"
####
[  0%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_Link.cpp.o
[  1%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_HotPoint.cpp.o
[  1%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_WayPoint.cpp.o
[  1%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_App.cpp.o
[  2%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_VirtualRC.cpp.o
[  2%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_HardDriver.cpp.o
[  2%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_API.cpp.o
[  3%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_Memory.cpp.o
[  3%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_Mission.cpp.o
[  3%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_Codec.cpp.o
[  4%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_Flight.cpp.o
[  4%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_Follow.cpp.o
[  4%] Building CXX object dji_sdk_lib/CMakeFiles/dji_sdk_lib.dir/src/DJI_Camera.cpp.o
Linking CXX static library /home/ubuntu/workspace/devel/lib/libdji_sdk_lib.a
[  4%] Built target dji_sdk_lib
[  4%] Building CXX object dji_sdk_read_cam/CMakeFiles/dji_sdk_read_cam.dir/src/nv_cam.cpp.o
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp: In function 'int main(int, char**)':
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:230:118: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.K = {388.40923066779754, 0.0, 318.06257844065226, 0.0, 518.1538449374815, 241.17339016626644, 0.0, 0.0, 1.0};
                                                                                                                      ^
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:230:13: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.K = {388.40923066779754, 0.0, 318.06257844065226, 0.0, 518.1538449374815, 241.17339016626644, 0.0, 0.0, 1.0};
             ^
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:231:59: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.R = {1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0};
                                                           ^
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:231:13: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.R = {1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0};
             ^
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:232:131: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.P = {373.5429992675781, 0.0, 317.51131336952494, 0.0, 0.0, 504.4360656738281, 240.6131009245937, 0.0, 0.0, 0.0, 1.0, 0.0};
                                                                                                                                   ^
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:232:13: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.P = {373.5429992675781, 0.0, 317.51131336952494, 0.0, 0.0, 504.4360656738281, 240.6131009245937, 0.0, 0.0, 0.0, 1.0, 0.0};
             ^
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:272:14: error: no match for 'operator=' (operand types are 'cv::Mat' and 'IplImage* {aka _IplImage*}')
    cvi.image = pImg;
              ^
/home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:272:14: note: candidates are:
In file included from /usr/local/include/opencv2/core/mat.hpp:3440:0,
                 from /usr/local/include/opencv2/core.hpp:59,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /opt/ros/indigo/include/cv_bridge/cv_bridge.h:43,
                 from /home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:3:
/usr/local/include/opencv2/core/mat.inl.hpp:575:6: note: cv::Mat& cv::Mat:perator=(const cv::Mat&)
Mat& Mat:perator = (const Mat& m)
      ^
/usr/local/include/opencv2/core/mat.inl.hpp:575:6: note:   no known conversion for argument 1 from 'IplImage* {aka _IplImage*}' to 'const cv::Mat&'
/usr/local/include/opencv2/core/mat.inl.hpp:3055:6: note: cv::Mat& cv::Mat:perator=(const cv::MatExpr&)
Mat& Mat:perator = (const MatExpr& e)
      ^
/usr/local/include/opencv2/core/mat.inl.hpp:3055:6: note:   no known conversion for argument 1 from 'IplImage* {aka _IplImage*}' to 'const cv::MatExpr&'
In file included from /usr/local/include/opencv2/core.hpp:59:0,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /opt/ros/indigo/include/cv_bridge/cv_bridge.h:43,
                 from /home/ubuntu/workspace/src/dji_sdk_read_cam/src/nv_cam.cpp:3:
/usr/local/include/opencv2/core/mat.hpp:1113:10: note: cv::Mat& cv::Mat:perator=(const Scalar&)
     Mat& operator = (const Scalar& s);
          ^
/usr/local/include/opencv2/core/mat.hpp:1113:10: note:   no known conversion for argument 1 from 'IplImage* {aka _IplImage*}' to 'const Scalar& {aka const cv::Scalar_<double>&}'
make[2]: *** [dji_sdk_read_cam/CMakeFiles/dji_sdk_read_cam.dir/src/nv_cam.cpp.o] Error 1
make[1]: *** [dji_sdk_read_cam/CMakeFiles/dji_sdk_read_cam.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 3 jobserver tokens available; should be 2!
Invoking "make -j2 -l2" failed

Could someone help me with that:

Thank you.

ItaloAW, 2016-08-01:
Dear package maintainer,

Please could you give any direction: This package doesn't works... I've already checked all the dependencies. Whats the problem with this:

andre-nguyen, 2016-08-01:
Just compile with the -j1 flag for now.

ItaloAW, 2016-08-01:
Thanks for the reply andre.

Unfortunately the error persists. 

> ubuntu@tegra-ubuntu:~/catkin_ws$ catkin_make --pkg dji_sdk_read_cam -j1    
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build
Devel space: /home/ubuntu/catkin_ws/devel
Install space: /home/ubuntu/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ubuntu/catkin_ws/build"
####
####
#### Running command: "make -j1" in "/home/ubuntu/catkin_ws/build/dji_sdk_read_cam"
####
[  5%] Building CXX object dji_sdk_read_cam/CMakeFiles/dji_sdk_read_cam.dir/src/nv_cam.cpp.o
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp: In function 'int main(int, char**)':
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:230:118: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.K = {388.40923066779754, 0.0, 318.06257844065226, 0.0, 518.1538449374815, 241.17339016626644, 0.0, 0.0, 1.0};
                                                                                                                      ^
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:230:13: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.K = {388.40923066779754, 0.0, 318.06257844065226, 0.0, 518.1538449374815, 241.17339016626644, 0.0, 0.0, 1.0};
             ^
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:231:59: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.R = {1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0};
                                                           ^
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:231:13: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.R = {1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0};
             ^
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:232:131: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.P = {373.5429992675781, 0.0, 317.51131336952494, 0.0, 0.0, 504.4360656738281, 240.6131009245937, 0.0, 0.0, 0.0, 1.0, 0.0};
                                                                                                                                   ^
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:232:13: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11 [enabled by default]
  cam_info.P = {373.5429992675781, 0.0, 317.51131336952494, 0.0, 0.0, 504.4360656738281, 240.6131009245937, 0.0, 0.0, 0.0, 1.0, 0.0};
             ^
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:272:14: error: no match for 'operator=' (operand types are 'cv::Mat' and 'IplImage* {aka _IplImage*}')
    cvi.image = pImg;
              ^
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:272:14: note: candidates are:
In file included from /usr/local/include/opencv2/core/mat.hpp:3440:0,
                 from /usr/local/include/opencv2/core.hpp:59,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /opt/ros/indigo/include/cv_bridge/cv_bridge.h:43,
                 from /home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:3:
/usr/local/include/opencv2/core/mat.inl.hpp:575:6: note: cv::Mat& cv::Mat::operator=(const cv::Mat&)
 Mat& Mat::operator = (const Mat& m)
      ^
/usr/local/include/opencv2/core/mat.inl.hpp:575:6: note:   no known conversion for argument 1 from 'IplImage* {aka _IplImage*}' to 'const cv::Mat&'
/usr/local/include/opencv2/core/mat.inl.hpp:3055:6: note: cv::Mat& cv::Mat::operator=(const cv::MatExpr&)
 Mat& Mat::operator = (const MatExpr& e)
      ^
/usr/local/include/opencv2/core/mat.inl.hpp:3055:6: note:   no known conversion for argument 1 from 'IplImage* {aka _IplImage*}' to 'const cv::MatExpr&'
In file included from /usr/local/include/opencv2/core.hpp:59:0,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /opt/ros/indigo/include/cv_bridge/cv_bridge.h:43,
                 from /home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:3:
/usr/local/include/opencv2/core/mat.hpp:1113:10: note: cv::Mat& cv::Mat::operator=(const Scalar&)
     Mat& operator = (const Scalar& s);
          ^
/usr/local/include/opencv2/core/mat.hpp:1113:10: note:   no known conversion for argument 1 from 'IplImage* {aka _IplImage*}' to 'const Scalar& {aka const cv::Scalar_<double>&}'
make[2]: *** [dji_sdk_read_cam/CMakeFiles/dji_sdk_read_cam.dir/src/nv_cam.cpp.o] Error 1
make[1]: *** [dji_sdk_read_cam/CMakeFiles/dji_sdk_read_cam.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1" failed
ubuntu@tegra-ubuntu:~/catkin_ws$ 

I've already reinstalled ROS and all the dependencies.

andre-nguyen, 2016-08-01:
Basically your error is
```
/home/ubuntu/catkin_ws/src/dji_sdk_read_cam/src/nv_cam.cpp:272:14: error: no match for 'operator=' (operand types are 'cv::Mat' and 'IplImage {aka IplImage}')
cvi.image = pImg;
```

Probably because this was written with Opencv 2.4 and you installed 3.0. Take a look at http://stackoverflow.com/questions/29636359/opencv-help-error-no-operator-matches-these-operands-operand-types-are-c

ItaloAW, 2016-08-08:
Andre,

Yes, the problem was a conflict with OpenCV deps. I've figured it out searching on ROS' forum, but my OpenCV version wasn't 3.0 and I've checked it many times. But yes, you are right... that was the problem, so I decided to download a Manifold new image and install everything again, but this time I didn't download any Ubuntu's update and the package was built successfully.  

I'd like to thank you for the support Andre! Thank you very much!

Best regards.

closed 2016-08-22 by amenonDJI