dji-sdk/Onboard-SDK-ROS #43: Attitude_control() function of ROS Onboard SDK to control M100?

cppman, 2016-08-01:
Am intending to develop a code to control Dji M100 in attitude: roll, pitch, thrust, and yaw. Using ROS Onboard sdk, attitude_control() would be the function to call. What is the syntax for this command,

attitude_control(FLAG, roll, pitch, thrust, yaw), right?

But it is not clear from the documentation on how to set FLAG, and the units of the different involved variables. Are the angles in rad or degree? Is the throttle in Newton or percentage?

lanyusea, 2016-08-01:
https://developer.dji.com/onboard-sdk/documentation/appendix/index.html

cppman, 2016-08-01:
@lanyusea I already seen that doc. But still it is clear 100% on how to write the command. Also please how to convert `0b00100xxy` to unsigned char variable?

lanyusea, 2016-08-01:
unsigned char flag = 0x00100011

lanyusea, 2016-08-01:
my fault... 0x means for bit per character. So it should be 0x20 or 0x21 or 0x22 or 0x23, depends on the value you set to x and y

cppman, 2016-08-01:
Thanks! Now more clear.
Please, can we we send throttle in Newton? What  is the unit for the angles?

lanyusea, 2016-08-01:
no
angle

cppman, 2016-08-01:
"no angle", what do you mean by that?

lanyusea, 2016-08-01:
>can we we send throttle in Newton? 
no

>What is the unit for the angles?
angle

cppman, 2016-08-01:
Thanks a lot!

cppman, 2016-08-02:
@lanyusea The drone does not seem taking into account the attitude control commands. Am reporting here the very short code am using:
`
int main(int argc, char **argv){

  ros::init(argc, argv, "m100ctrl");
  ROS_INFO("sdk_service_client_test");
  ros::NodeHandle nh;
  DJIDrone* drone = new DJIDrone(nh);

  cout << "requesting sdk permission control... " << endl;
  drone->request_sdk_permission_control();
  
  
  cout << "Take off... " << endl;
  drone->takeoff();
  usleep(12*1e6);

  drone->release_sdk_permission_control();
    
  while (ros::ok()){
    
    double roll(0), pitch(0), thr(90), yaw(0);

    for(int  i=0; i<5000; ++i){
      cout << "***** Attitude control " << i << endl;
      drone->request_sdk_permission_control();
      drone->attitude_control(0x2A, (float)roll, (float)pitch, (float)thr, (float)yaw);
      usleep(1e3);
    }

   
  }

  cout << "Landing...!!!!!!!!!!!!!!!!!!!!!! " << endl<< endl<< endl<< endl;
    drone->landing();
    
  cout << "... end."<< endl;

  // ros::spin();
  return 0;
}
`

lanyusea, 2016-08-02:
you don't need to release the control after taking off while request it again before each attitude control.

I suggest use usleep(20000) for 50Hz

the difference is just `stable` and `not stable`, as it is... :ghost:

cppman, 2016-08-02:
Still the same issue though I remove  `drone->release_sdk_permission_control();` and `drone->request_sdk_permission_control();`. It takes a long time (more than one minute) to reach a high value of thrust; conclusion with the naked-eye.

By the way, is there a method to know the command the drone is actually applying?

amenonDJI, 2016-08-22:
@cppman Could you elaborate your setup? Are you flying the aircraft to test this or are you on a simulator?

closed 2016-12-01 by amenonDJI

albert21martins, 2017-04-14:
hi @cppman @amenonDJI ,

did you get it running? I am having the same issue. Where are developing a similar application. We are able to run in simulator but once going outside and  tried in a real copter with A3 we can takeoff and land. For example the attitude control sample runs just take off and nothing else. 

Can you please share some information about it?

voitgxd, 2017-07-18:
hi @albert21martins I am having the same issur. did you get it running in the realword. I am also able to run it in simulator but failed in outside.

Can you please share some infomation about it?

albert21martins, 2017-07-18:
Hi! Yes I got it running, but to be sincere I do not know how did we do. We had 2xA3, one was connected in the simulator, the other one was in a copter. Since the simulator worked but copter did not, I copied all the A3 configs from one to another. This way worked for me. So I will assume the problems come from A3-API configurations.
Hope to be a little helpfull

voitgxd, 2017-07-19:
@albert21martins Can you show me your API configuration? I am going to compare it with mine to find some thing error.

koyahatataku, 2019-01-04:
I want to see @albert21martins ' API configuration, too !

koyahatataku, 2019-01-16:
@albert21martins 
My setings is like below:
[image]
I want to know the differences between mine and yours.