dji-sdk/Onboard-SDK-ROS #45: Dji M100 autopilot commands not taken into account

cppman, 2016-08-18:
Am trying to control dji M100 in attitude with Onboard-SDK-ROS/dji_sdk. Blow is the simple and very short code am using:

```
#include <dji_sdk/dji_drone.h>
#include <ros/ros.h>
#include <unistd.h>

using namespace std;

int main(int argc, char **argv){

  ros::init(argc, argv, "m100ctrl");
  ROS_INFO("sdk_service_client_test");
  ros::NodeHandle nh;
  DJIDrone* drone = new DJIDrone(nh);
  ros::Rate rate(50.);
  
  cout << "requesting sdk permission control... " << endl;
  drone->request_sdk_permission_control();
  //drone->release_sdk_permission_control();
  
  cout << "Take off... " << endl;
  drone->takeoff();
  usleep(5*1e6); // in sec
  cout << endl <<  endl << "**** End of take off... " << endl;

  
  //************** LOOP
  //drone->release_sdk_permission_control();
  int itr(0);
  while (ros::ok() && itr <1000)
    {
      double roll(0), pitch(0), thr(90), yaw(0);
      cout << "***** Attitude control " << itr << endl;
      // drone->request_sdk_permission_control();
      drone->attitude_control(0x2A /*2A*/, (float)roll, (float)pitch, (float)thr, (float)yaw);
      ros::spinOnce();
      ++itr;
      rate.sleep();
    }

  cout << "Landing... " << endl<< endl<< endl<< endl;
  drone->landing();
    
  cout << "... end."<< endl;

  
  return 0;
}
```

Takeoff is taken into account. However, the rest of the code 

 `drone->attitude_control(0x2A /*2A*/, (float)roll, (float)pitch, (float)thr, (float)yaw);`

does not seem at all taken correctly int account. The drone behaves erratically. Sometimes the motors tun with low speed, while sometimes they slowly accelerate (sometimes several minutes) to reach high speed.

Why this? Am I missing something?Am really stuck at this point....

amenonDJI, 2016-08-22:
@cppman I would suggest you to try with Control Flag 0x22 rather than 0x2A. With 0x2A you are commanding Yaw rate not yaw angle. Here is the [reference](https://developer.dji.com/onboard-sdk/documentation/appendix/index.html#flight-data). 

Also, I would suggest you try this on a simulator first before you try it on a real aircraft.

cppman, 2016-08-23:
I tried with 0x22 flag but the same strange drone behavior... Sometimes it slowly accelerates and other time the motors turn at low speed even if I put thrust command to max (100).

Perhaps that the ROS Onboard sdk does not work properly (?)

AlexandreBorowczyk, 2016-09-12:
@cppman. Are you testing on the dji simulator? Also when you are saying the motors are slowly accelerate sometimes several minutes, did you change the number of iteration of the while loop? 
Normaly 20 sec at 90% thrust would send the m100 over 40m.

amenonDJI, 2016-10-19:
@cppman Have you had a chance to try the Linux sample with Attitude control mode? It is easy to use and run. Modify one of the functions to send commands in attitude mode. If you still have problems, please report back.

closed 2016-12-01 by amenonDJI