dji-sdk/Onboard-SDK-ROS #49: [Problem] DJI SDK-ROS Mavlink example

won13y, 2016-09-07:
wkyoun@wkyoun:~$ roslaunch dji_sdk_dji2mav dji2mav_bringup.launch 

    ... logging to /home/wkyoun/.ros/log/e7964014-7499-11e6-85c9-00c2c6cf6a1b/roslaunch-wkyoun-    7686.log
    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.

    started roslaunch server http://192.168.43.124:36496/

    SUMMARY
    ========

    PARAMETERS
     * /dji2mav_bringup/srcPort1: 14555
     * /dji2mav_bringup/srcPort2: 14549
     * /dji2mav_bringup/targetIp1: 192.168.43.138
     * /dji2mav_bringup/targetIp2: 192.168.43.138
     * /dji2mav_bringup/targetPort1: 14550
     * /dji2mav_bringup/targetPort2: 14551
     * /rosdistro: indigo
     * /rosversion: 1.11.20

    NODES
      /
        dji2mav_bringup (dji_sdk_dji2mav/dji2mav_bringup)

    ROS_MASTER_URI=http://192.168.43.124:11311

    core service [/rosout] found
    process[dji2mav_bringup-1]: started with pid [7704]
    [dji2mav] [INFO] Going to establish connection for Manager with index 0. The size of sender list is     set to 16, the size of send buf and recv buf are set to 512 and 4096...
    [dji2mav] [INFO] Going to establish connection with 192.168.43.138:14550 using port 14555...
    [dji2mav] [INFO] Attempt to connect to 192.168.43.138:14550 using local port 14555...
    [dji2mav] [INFO] ...connection succeed.
    [dji2mav] [INFO] ...succeed in establishing connection.
    [dji2mav] [INFO] ...succeed in establishing connection for Manager 0.
    [dji2mav] [INFO] Going to establish connection for Manager with index 1. The size of sender list is     set to 16, the size of send buf and recv buf are set to 512 and 4096...
    [dji2mav] [INFO] Going to establish connection with 192.168.43.138:14551 using port 14549...
    [dji2mav] [INFO] Attempt to connect to 192.168.43.138:14551 using local port 14549...
    [dji2mav] [INFO] ...connection succeed.
    [dji2mav] [INFO] ...succeed in establishing connection.
    [dji2mav] [INFO] ...succeed in establishing connection for Manager 1.
    [dji2mav] [INFO] Succeed in registering module heartbeat.
    [dji2mav] [INFO] Succeed in registering module sensors.
    [dji2mav] [INFO] Succeed in registering module waypoint.
    [dji2mav] [INFO] Succeed in registering module hotpoint.
    [dji2mav_bringup-1] process has died [pid 7704, exit code -11, cmd /home/wkyoun/catkin_ws/devel    /lib/dji_sdk_dji2mav/dji2mav_bringup __name:=dji2mav_bringup __log:=/home/wkyoun    /.ros/log/e7964014-7499-11e6-85c9-00c2c6cf6a1b/dji2mav_bringup-1.log].
    log file: /home/wkyoun/.ros/log/e7964014-7499-11e6-85c9-00c2c6cf6a1b/dji2mav_bringup-1*.log
    all processes on machine have died, roslaunch will exit
    shutting down processing monitor...
    ... shutting down processing monitor complete
    done

Warning: You have connected QGroundControl to a firmware flight stack which is only partially     supported. If you are using the APM Flight Stack it is currently recommended to use Mission Planner, APM Planner or Tower as your ground control station.

[image]

pcler, 2016-09-07:
Would you please provide the version of QGroundStation? We will do some further check inside

won13y, 2016-09-07:
I have tried QGroundStation 2.9.7, and 2.7.1(Stable) version,
QGroundStation 2.9.7 results in the sound output described in above
The output sound of Qgroundcontrol is "communication retained" and right after "communication lost"

won13y, 2016-09-20:
@pcler 

I conclude that above problem is not due to QGroundStation, but DJI SDK is the problem itself.

The reason why I conclude in this way is as followings

1) dji2mav package is not working regardless of running QGroundStation as following
    
   **dji2mav package results in  shutting down processing as following**
 
     dji2mav package should not be stopped as following regardless of running QGroundStation if  dji2mav package is correct
 
    wkyoun@wkyoun:~/catkin_ws/src/Onboard-SDK-ROS$ roslaunch dji_sdk_dji2mav dji2mav_bringup.launch 
    ... logging to /home/wkyoun/.ros/log/b3729030-7f07-11e6-8328-b8aeedea2130/roslaunch-wkyoun-2798.log
    Checking log directory for disk usage. This may take awhile.
    Press Ctrl-C to interrupt
    Done checking log file disk usage. Usage is <1GB.

    started roslaunch server http://192.168.0.10:40641/

    SUMMARY
    ========

    PARAMETERS
     * /dji2mav_bringup/srcPort1: 14551
     * /dji2mav_bringup/srcPort2: 14555
     * /dji2mav_bringup/targetIp1: 192.168.0.9
     * /dji2mav_bringup/targetIp2: 192.168.0.11
     * /dji2mav_bringup/targetPort1: 14550
     * /dji2mav_bringup/targetPort2: 14554
     * /rosdistro: indigo
     * /rosversion: 1.11.19

    NODES
      /
        dji2mav_bringup (dji_sdk_dji2mav/dji2mav_bringup)

    ROS_MASTER_URI=http://192.168.0.10:11311

    core service [/rosout] found
    process[dji2mav_bringup-1]: started with pid [2816]
    [dji2mav] [INFO] Going to establish connection for Manager with index 0. The size of sender list is set to 16, the size of send buf and recv buf are set to 512 and 4096...
    [dji2mav] [INFO] Going to establish connection with 192.168.0.9:14550 using port 14551...
    [dji2mav] [INFO] Attempt to connect to 192.168.0.9:14550 using local port 14551...
    [dji2mav] [INFO] ...connection succeed.
    [dji2mav] [INFO] ...succeed in establishing connection.
    [dji2mav] [INFO] ...succeed in establishing connection for Manager 0.
    [dji2mav] [INFO] Going to establish connection for Manager with index 1. The size of sender list is set to 16, the size of send buf and recv buf are set to 512 and 4096...
    [dji2mav] [INFO] Going to establish connection with 192.168.0.11:14554 using port 14555...
    [dji2mav] [INFO] Attempt to connect to 192.168.0.11:14554 using local port 14555...
    [dji2mav] [INFO] ...connection succeed.
    [dji2mav] [INFO] ...succeed in establishing connection.
    [dji2mav] [INFO] ...succeed in establishing connection for Manager 1.
    [dji2mav] [INFO] Succeed in registering module heartbeat.
    [dji2mav] [INFO] Succeed in registering module sensors.
    [dji2mav] [INFO] Succeed in registering module waypoint.
    [dji2mav] [INFO] Succeed in registering module hotpoint.
    [dji2mav_bringup-1] process has died [pid 2816, exit code -11, cmd /home/wkyoun/catkin_ws/devel/lib/dji_sdk_dji2mav/dji2mav_bringup __name:=dji2mav_bringup __log:=/home/wkyoun/.ros/log/b3729030-7f07-11e6-8328-b8aeedea2130/dji2mav_bringup-1.log].
    log file: /home/wkyoun/.ros/log/b3729030-7f07-11e6-8328-b8aeedea2130/dji2mav_bringup-1*.log
    all processes on machine have died, roslaunch will exit
    shutting down processing monitor...
    ... shutting down processing monitor complete
    done

2) dji2mav package is running as following when I delete 263 line(container.run()) in dji2mav_bringup.cpp file without even running QGroundStation.

https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.1/dji_sdk_dji2mav/src/dji2mav_bringup.cpp

    wkyoun@wkyoun:~/catkin_ws/src/Onboard-SDK-ROS$ roslaunch dji_sdk_dji2mav dji2mav_bringup.launch 

[... 32 lines cut ...]
    [dji2mav] [INFO] ...connection succeed.
    [dji2mav] [INFO] ...succeed in establishing connection.
    [dji2mav] [INFO] ...succeed in establishing connection for Manager 0.
    [dji2mav] [INFO] Going to establish connection for Manager with index 1. The size of sender list is set to 16, the size of send buf and recv buf are set to 512 and 4096...
    [dji2mav] [INFO] Going to establish connection with 192.168.0.11:14554 using port 14555...
    [dji2mav] [INFO] Attempt to connect to 192.168.0.11:14554 using local port 14555...
    [dji2mav] [INFO] ...connection succeed.
    [dji2mav] [INFO] ...succeed in establishing connection.
    [dji2mav] [INFO] ...succeed in establishing connection for Manager 1.
    [dji2mav] [INFO] Succeed in registering module heartbeat.
    [dji2mav] [INFO] Succeed in registering module sensors.
    [dji2mav] [INFO] Succeed in registering module waypoint.
    [dji2mav] [INFO] Succeed in registering module hotpoint.

It seem to me that 

https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.1/dji_sdk_dji2mav/src/dji2mav_bringup.cpp

above code handles simultaneously two  QGroundStation.

Would you please modify above source code to handle only 1 GCS(QGroundStation)?

Please resolve aforementioned problem ASAP since we have been waiting a long time!

I know that a new GCS for A3 will be release soon but we really need QGroundStation

yguo18, 2017-01-04:
I  want  to  know  how  to  set  ip and port?   what does ip and port concrete meaning

hgvicban, 2018-01-09:
@won13y Were you able to get past this issue? I have updated dji2mav_bringup.cpp to handle single GCS only as per my requirement but same issue still occurs - connection succeed but few seconds later processes die.

Desync-o-tron, 2019-03-24:
@wkyoun 

Did you have any luck with this?
I've come to the same conclusion that there is something wrong in the code of dji2mav_bringup.cpp.
I'm getting the same errors as you.

Also, were you trying to execute this code on the jetson?

dji-kyle, 2021-06-25:
Please update to OSDK 3.9 + ROS 3.8,
The following forum community is recommanded for you, if the problem persists, or new questions about the DJI SDK,
https://djisdksupport.zendesk.com/hc/en-us/community/topics

closed 2021-06-25 by dji-kyle