dji-sdk/Onboard-SDK-ROS #50: [Problem] dji_sdk_dji2mav crashes at start

diegocepedaw, 2016-10-07:
After launching with command  roslaunch dji_sdk_dji2mav dji2mav_bringup.launch 
I get the following:

started roslaunch server http://diego-UX303UB:33038/

SUMMARY
========

PARAMETERS
 * /dji2mav_bringup/srcPort1: 14551
 * /dji2mav_bringup/srcPort2: 14555
 * /dji2mav_bringup/targetIp1: 127.0.0.1
 * /dji2mav_bringup/targetIp2: 127.0.0.1
 * /dji2mav_bringup/targetPort1: 14550
 * /dji2mav_bringup/targetPort2: 14554
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    dji2mav_bringup (dji_sdk_dji2mav/dji2mav_bringup)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[dji2mav_bringup-1]: started with pid [17272]
[dji2mav] [INFO] Going to establish connection for Manager with index 0. The size of sender list is set to 16, the size of send buf and recv buf are set to 512 and 4096...
[dji2mav] [INFO] Going to establish connection with 127.0.0.1:14550 using port 14551...
[dji2mav] [INFO] Attempt to connect to 127.0.0.1:14550 using local port 14551...
[dji2mav] [INFO] ...connection succeed.
[dji2mav] [INFO] ...succeed in establishing connection.
[dji2mav] [INFO] ...succeed in establishing connection for Manager 0.
[dji2mav] [INFO] Going to establish connection for Manager with index 1. The size of sender list is set to 16, the size of send buf and recv buf are set to 512 and 4096...
[dji2mav] [INFO] Going to establish connection with 127.0.0.1:14554 using port 14555...
[dji2mav] [INFO] Attempt to connect to 127.0.0.1:14554 using local port 14555...
[dji2mav] [INFO] ...connection succeed.
[dji2mav] [INFO] ...succeed in establishing connection.
[dji2mav] [INFO] ...succeed in establishing connection for Manager 1.
[dji2mav] [INFO] Succeed in registering module heartbeat.
[dji2mav] [INFO] Succeed in registering module sensors.
[dji2mav] [INFO] Succeed in registering module waypoint.
[dji2mav] [INFO] Succeed in registering module hotpoint.
[dji2mav_bringup-1] process has died [pid 17272, exit code -11, cmd /home/diego/catkin_ws/devel/lib/dji_sdk_dji2mav/dji2mav_bringup __name:=dji2mav_bringup __log:=/home/diego/.ros/log/a6f963ec-8cd6-11e6-8dde-1002b5ee16c8/dji2mav_bringup-1.log].
log file: /home/diego/.ros/log/a6f963ec-8cd6-11e6-8dde-1002b5ee16c8/dji2mav_bringup-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I tried using another computer instead of localhost and got the same result
My .launch looks like this

<launch>

    <arg name="targetIp1" default="127.0.0.1" />
    <arg name="targetPort1" default="14550" />
    <arg name="srcPort1" default="14551" />

    <arg name="targetIp2" default="127.0.0.1" />
    <arg name="targetPort2" default="14548" />
    <arg name="srcPort2" default="14549" />

    <node name="dji2mav_bringup" pkg="dji_sdk_dji2mav" type="dji2mav_bringup" output="screen">
        <param name="targetIp1" type="string" value="$(arg targetIp1)" />
        <param name="targetPort1" type="int" value="$(arg targetPort1)" />
        <param name="srcPort1" type="int" value="$(arg srcPort1)" />

        <param name="targetIp2" type="string" value="$(arg targetIp2)" />
        <param name="targetPort2" type="int" value="$(arg targetPort2)" />
        <param name="srcPort2" type="int" value="$(arg srcPort2)" />
    </node>

</launch>

spacemig, 2016-12-06:
I'm having a similar issue running on the Raspberry PI. Any thoughts on how to resolve this?

```
pi@raspberrypi:~/ros_catkin_ws $ roslaunch dji_sdk_dji2mav dji2mav_bringup.launch
... logging to /home/pi/.ros/log/5d4d6dc2-bb5e-11e6-9ab7-b827eba2bbbb/roslaunch-raspberrypi-9969.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:45015/

SUMMARY
========

PARAMETERS
 * /dji2mav_bringup/srcPort1: 14551
 * /dji2mav_bringup/srcPort2: 14549
 * /dji2mav_bringup/targetIp1: 192.168.152.200
 * /dji2mav_bringup/targetIp2: 192.168.152.179
 * /dji2mav_bringup/targetPort1: 14550
 * /dji2mav_bringup/targetPort2: 14550
 * /rosdistro: kinetic
 * /rosversion: 1.12.6

NODES
  /
    dji2mav_bringup (dji_sdk_dji2mav/dji2mav_bringup)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[dji2mav_bringup-1]: started with pid [9987]
[dji2mav] [INFO] Going to establish connection for Manager with index 0. The size of sender list is set to 16, the size of send buf and recv buf are set to 512 and 4096...
[dji2mav] [INFO] Going to establish connection with 192.168.152.200:14550 using port 14551...
[dji2mav] [INFO] Attempt to connect to 192.168.152.200:14550 using local port 14551...
[dji2mav] [INFO] ...connection succeed.
[dji2mav] [INFO] ...succeed in establishing connection.
[dji2mav] [INFO] ...succeed in establishing connection for Manager 0.
[dji2mav] [INFO] Going to establish connection for Manager with index 1. The size of sender list is set to 16, the size of send buf and recv buf are set to 512 and 4096...
[dji2mav] [INFO] Going to establish connection with 192.168.152.179:14550 using port 14549...
[dji2mav] [INFO] Attempt to connect to 192.168.152.179:14550 using local port 14549...
[dji2mav] [INFO] ...connection succeed.
[dji2mav] [INFO] ...succeed in establishing connection.
[dji2mav] [INFO] ...succeed in establishing connection for Manager 1.
[dji2mav] [INFO] Succeed in registering module heartbeat.
[dji2mav] [INFO] Succeed in registering module sensors.
[dji2mav] [INFO] Succeed in registering module waypoint.
[dji2mav] [INFO] Succeed in registering module hotpoint.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [INFO] No datagram received in last 10 sec.
[dji2mav] [WARN] In file: /home/pi/ros_catkin_ws/src/Onboard-SDK-ROS/dji_sdk_dji2mav/include/dji_sdk_dji2mav/mavHandler.h, function: updateSysid, line: 183. Switch sysid of Manager with index 0 from 0 to 255.
[dji2mav_bringup-1] process has died [pid 9987, exit code -11, cmd /opt/ros/kinetic/lib/dji_sdk_dji2mav/dji2mav_bringup __name:=dji2mav_bringup __log:=/home/pi/.ros/log/5d4d6dc2-bb5e-11e6-9ab7-b827eba2bbbb/dji2mav_bringup-1.log].
log file: /home/pi/.ros/log/5d4d6dc2-bb5e-11e6-9ab7-b827eba2bbbb/dji2mav_bringup-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

chutsu, 2017-02-19:
its hard to debug with that, I would suggest running `gdb` with the ros node by adding a `launch-prefix` flag to the node in the launch file. Look [here](http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB) for more info

Desync-o-tron, 2019-04-01:
Howdy! I fixed this by running it on a Jetson TX2 - my x86 laptop would crash as you've shown above. Both running ubuntu 16.

closed 2021-06-25 by dji-kyle