dji-sdk/Onboard-SDK-ROS #51: compatibility with Matrice 600

miguelolivaresmendez, 2016-10-19:
I planing to buy a DJI Matrice 600 for research purpose, because I need more payload. I know that the onboard SDK is compatible with Matrice 600, what about the ROS version? is it compatible or is planned to be compatible? Is the manifold compatible with the Matrice 600? If so, me and my team will participate actively in this development.
Many thanks in advanced

amenonDJI, 2016-12-01:
Yes the M600 does support the Onboard SDK pure Linux version and ROS versions. The manifold is also compatible with the M600. 

Please re-open the case if you have any other technical questions.

closed 2016-12-01 by amenonDJI

gusmally, 2017-03-04:
Hi, will the Manifold 200 be compatible with the Manifold?