dji-sdk/Onboard-SDK-ROS #52: Matrice 100 can't connect to a second computer

robotjackie, 2016-10-31:
Help! We connected our group's M100 to a laptop, and it seems to be "bound" to that computer. No one else's computer seems to be able to connect to our M100. Please advise - is there a way around this? Is this a bug? Is there a way to "reset" the connection so a different computer can connect to it?

andre-nguyen, 2016-10-31:
Did you write the correct API key in the other computer?

robotjackie, 2016-10-31:
Yes I did! As well as the correct drone ID number.

-- 
Jackie Wu
Founder | Eighty Nine Robotics
<http://t.sidekickopen66.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJW7t5XZs3MPjkCW8qSJ-23M2f9xVd75C256dyRpf4b5b2n02?t=http%3A%2F%2Fwww.eightyninerobotics.com%2F&si=5636171506647040&pi=f32c7df2-0758-4978-fd5e-638ff32f8b62>
512-573-9091

------------------------------
This email and any files transmitted with it are confidential and intended
solely for the use of the individual or entity to whom they are addressed.
Please notify the sender immediately by e-mail if you have received this
e-mail by mistake and delete this e-mail from your system.

andre-nguyen, 2016-10-31:
What's the console output

AlexandreBorowczyk, 2016-11-01:
Did you have the remote control powered on with a cellphone with DJIGo open?
I think sometime the drone need to authenticate a new connection through the app.

robotjackie, 2016-11-01:
Thanks guys: http://forum.dev.dji.com/thread-32671-1-1.html

amenonDJI, 2016-11-01:
@robotjackie Could you tell us what fixed the issue? Was it a problem with not configuring your port or something else? Your solution will help users in the future who run into the same problem.

robotjackie, 2016-11-01:
Sorry, to clarify, the issue is NOT fixed. We have no idea why it still
won't connect. I was posting the link as a response to requests for what we
were saying, and saying "thanks" in advance for any insight or advice
offered!

-- 
Jackie Wu
Founder | Eighty Nine Robotics
<http://t.sidekickopen66.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJW7t5XZs3MPjkCW8qSJ-23M2f9xVd75C256dyRpf4b5b2n02?t=http%3A%2F%2Fwww.eightyninerobotics.com%2F&si=5636171506647040&pi=b8da692c-85ea-4613-88c1-468c7724fd70>
512-573-9091

------------------------------
This email and any files transmitted with it are confidential and intended
solely for the use of the individual or entity to whom they are addressed.
Please notify the sender immediately by e-mail if you have received this
e-mail by mistake and delete this e-mail from your system.

amenonDJI, 2016-11-01:
@robotjackie Could you check your connection. If you only see "succeed to create thread,Broadcast call back received" means your computer is not communicating with the drone. Sometimes this can happen if the Rx and Tx on the USB-UART line was switched.

robotjackie, 2016-11-01:
I don't believe that's the problem, because with the line unchanged, we
connect fine to the first laptop...can fly it, control it, etc. But then
when we connect via USB to a second or third computer, we get the problem.

rohitsantdji, 2016-11-01:
Can you successfully activate it and run it using the linux sample on multiple computers?

robotjackie, 2016-11-01:
Hi Rohit,
Could you rephrase the question? I'm not completely sure what you mean. Our
problem is that we CAN'T connect over multiple Linux computers - we can
only connect with one, and regardless of the proper key and credentials
provided, no other computer we have can connect to our Matrice 100!

rohitsantdji, 2016-11-01:
[This](https://developer.dji.com/onboard-sdk/documentation/github-platform-docs/Linux/README.html) is the linux sample program I was referring to, the one that does not use ROS. Running the linux sample will help get us another data point on the SDK communication with the drone.

Follow the guide [here](https://developer.dji.com/onboard-sdk/documentation/github-platform-docs/Linux/README.html#setup) to get set up with the sample. You will have to clone the non-ROS [Onboard SDK](https://github.com/dji-sdk/Onboard-SDK) github repo. If you are running this on something other than a Manifold or an x86-based computer, please checkout the 3.1.8 tag on the repo and build the Linux sample using the following instructions: 

* First, we need to create the directories `objs` and `bin` inside the `sample/Linux` directory. To do this, open up a terminal inside the `sample/Linux` directory and type `mkdir bin; mkdir objs` at the command line. 
* Next, type `make` at the command prompt.
* To access the serial port, add your username to the dialout group by typing `sudo usermod -a -G dialout $USER` (you do not need to replace $USER with your username). **Then logout and login again.**  

As a side note, try the last step above on your ROS setup too if you haven't changed the ownership of your serial port on your other machines.

desmosedic, 2016-11-02:
@robotjackie I came across one more scenario where you can experience the problem you're seeing. When you have the second computer hooked up to the M100's UART port, do you have the first (successfully activated) computer still hooked up to the M100's other UART port? Even if the first computer is simply hooked up without running the OSDK, it results in the console output stopping at "succeed to create thread,Broadcast call back received". Only one computer must be connected to aircraft's UART port at any time.

amenonDJI, 2016-11-02:
@robotjackie  I came across one more scenario where you can experience the problem you're seeing. When you have the second computer hooked up to the M100's UART port, do you have the first (successfully activated) computer still hooked up to the M100's other UART port? Even if the first computer is simply hooked up without running the OSDK, it results in the console output stopping at "succeed to create thread,Broadcast call back received". Only one computer must be connected to aircraft's UART port at any time.

robotjackie, 2016-11-02:
Thanks Arjun for the advice! Unfortunately, that's not the case. We unplug
everything and move it to the second computer, with no connection to the
first.
Out of curiosity, have you/anyone else been able to successfully connect
the Matrice to multiple computers?

amenonDJI, 2016-11-02:
@robotjackie Yes, our setup frequently requires multiple computers connecting to the same M100.

robotjackie, 2016-11-02:
Dang it! That's awesome. I wonder what's wrong with ours....
I wonder if there's a way for them to "re-set" the connection, from DJI's
servers' side.

-- 
Jackie Wu
Founder | Eighty Nine Robotics
<http://t.sidekickopen66.com/e1t/c/5/f18dQhb0S7lC8dDMPbW2n0x6l2B9nMJW7t5XZs3MPjkCW8qSJ-23M2f9xVd75C256dyRpf4b5b2n02?t=http%3A%2F%2Fwww.eightyninerobotics.com%2F&si=5636171506647040&pi=09d2101d-ad32-439e-beba-5abe698d24ba>
512-573-9091

------------------------------
This email and any files transmitted with it are confidential and intended
solely for the use of the individual or entity to whom they are addressed.
Please notify the sender immediately by e-mail if you have received this
e-mail by mistake and delete this e-mail from your system.

amenonDJI, 2016-11-02:
@robotjackie As @lanyusea mentioned on the developer forum, there isn't any lock/set feature on the server side for this to happen. It needs to be debugged on your side. As @rohitsantdji mentioned, a good test would be to try out the Linux sample.

closed 2016-12-01 by amenonDJI

YorksonChang, 2017-02-26:
@robotjackie hey, I wanna know have you solved this problem? 'Cause I'm faced with this, too. I have connect one Linux Computer with M100 but failed to connect a second Linux Computer.
Thanks guys:
https://github.com/dji-sdk/Onboard-SDK-ROS/issues/68

abdullah38rcc, 2017-03-09:
Hello everyone. I have the same problem. I am using Jetson Tegra board tk1. It is running 14.04 ubuntu. I made a successful connection using Ubuntu 16.04. But when i have tried to connect with my Jetson board, it showed broadcast call received. Didn't activate. 
Now, after upgrading firmware to the last version, It is showing getDrone version error !! Tried everything and still it is showing the same error !! Is it happening for update error ? Please anyone answer it. I need urgent solution. Now feeling so hopeless. Any kind of help is appreciated. I tried command line even. With this new update nothing is working !! Please someone look int it or DJI developers please check it.

amenonDJI, 2017-03-09:
@abdullah38rcc Did you get the latest hotfix from the OSDK repo dated February 3rd? Are you also using an M100? This is a different issue from what the OP here was having. I'd suggest opening a new issue.

abdullah38rcc, 2017-03-10:
@amenonDJI  Yes, I'm using an M100. It used to work before the last update. I have been developing our code for an algorithm test. Firstly, the issue was same for us. It worked fine with same pc having windows 10 and ubuntu 16.04 with ROS kinetic installed. USB port was '/dev/ttyUSB0' for them. But when i have used Nvidia Jetson tk1 board for our project it has Ubuntu 14.04 installed with ROS indigo, it showed us following line: Stuck in 'Broadcast callback received'  here is a detail issue:

http://forum.dev.dji.com/thread-32671-1-1.html 

Secondly, we tried to solve the USB port issue with setting up port '/dev/ttyTHS1'. That time is the February month. After that we  updated with latest version of OSDK. Now, ROS it is showing another problem: ERROR activate,line 388: Please call getDroneVersion first 

read detail issue :

https://github.com/dji-sdk/Onboard-SDK-ROS/issues/68

We are having this activation problem now !! Even my pc which worked before is now showing same error with the new update. I have tried with 3.1 and 2.3 OSDK also, nothing worked. Ending up with same error. In addition, we do not have the camera, so during new update we faced some problem but finally updated the drone version.
   
Please suggest me a solution. We are preparing for a research project submission and near deadline having such issues unsolved. We need your serious help to overcome it.