dji-sdk/Onboard-SDK-ROS #54: A3 takeoff service not working

DARCLab, 2016-11-24:
I am unable to take off from the dji_sdk_demo while running in the dji's assistant 2's simulator.

lanyusea, 2016-11-24:
are your working on the latest A3 firmware?

DARCLab, 2016-11-28:
The firmware tab in DJI assistant 2 won't show me the version number, but I updated to DJI Assistant 2 v1.0.6 and it tells me that the A3 firmware needs to be updated. So correct me if I am wrong but the latest firmware for the A3 is v1.3.0.0, so I am working on the previous version of the A3 firmware which would be v1.2.0.0.

DARCLab, 2016-11-28:
Also I am unable to test the lastest A3 firmware because both DJI Assistant 2 v1.0.5 or v1.0.6 won't let me upgrade the firmware.  It just tells me to retry.  Is anybody else having this issue?

lanyusea, 2016-11-29:
It seems that you meet some technical problem with A3, I suggest you
contact the DJI after-sale support team.

closed 2016-12-01 by amenonDJI