dji-sdk/Onboard-SDK-ROS #55: DJI on board SDK ROS, only fly to the first way point and the go home

eastmoutain, 2016-12-04:
`   // something to prepare the autopilot
    newWaypointList.waypoint_list.push_back(waypoint1);
    newWaypointList.waypoint_list.push_back(waypoint2);
    newWaypointList.waypoint_list.push_back(waypoint3); 
    newWaypointList.waypoint_list.push_back(waypoint4);

    drone->waypoint_navigation_send_request(newWaypointList);
    drone->gohome(); `

   both in the simulator or the true environment, it only goes to the first point and then return home, is there any problem of my code? I really can't figure it out!

lanyusea, 2016-12-05:
what will happen if you remove the gohome()?

eastmoutain, 2016-12-05:
sorry, i haven't tried removing the the gohome() sentence,  but i don't think that it was caused by the gohome() sentence. 

I did the experiment on M600, and do a little change according to DJI SDK ROS client app as show above, just add four points to the way point list. Then i tried it both on simulator and outside of door, the results are same.

you can try it on your platform, frankly speaking, it is not convenient for me to test because i can't touch my M600 at present.

lanyusea, 2016-12-05:
well, I understand your meaning, however, in fact. The `waypoint_navigaiton_send_request` is not the waypoint function in drone's controller but an implementation in ROS based on the actionLib. 

You can check the [code](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.1/dji_sdk/src/modules/dji_sdk_node_actions.cpp), it is something like:

```c
for(not reach the last one)
    while(trying to reach the current waypoint)
         do velocity control according to how many distance left
    done
done
```

and then you send another command to homing after that.

There should be no problem if our system is in serial. However, unfortunately, we implemented the system by multi-thread.

I'm not familiar with ROS, but from my undertanding:
we created four threads by `ros::AsyncSpinner spinner(4);`, what's worse, they are non-blocking parallel threads, which means they won't block each other when processing callbacks.

so it is possible that while one thread is controlling the drone reaching your waypoint, the other one send a homing command to the drone, which abandoned the waypoint task.

So to solve:

the actionLib has a topic called `progress` to publish if the work is done or not. You can call the homing after all waypoint tasks are finished.

OR

change the system into 

OR

use the waypoint API provided by the drone API, which is the case 22 in `client.cpp`. (remember to send a command to start your task after uploading waypoint task)

http://wiki.ros.org/roscpp/Overview/Callbacks%20and%20Spinning
http://answers.ros.org/question/53055/ros-callbacks-threads-and-spinning/

eastmoutain, 2016-12-06:
well i think i should have a try to direct the autopilot fly the way points one by one according to your suggestion that polling the autopilot status or position.

how can I get (or subscribe) the progress feed back info which is published by 
`waypoint_navigation_action_server->publishFeedback(waypoint_navigation_feedback);`
is there any API for this function?

lanyusea, 2016-12-06:
you can subscribe the topic `dji_sdk/WaypointNavigationFeedback`

AldrichCabrera, 2017-01-26:
When the waypoints test is launched, a baidu map is linked when connected to the Matrice 100. Can the map be changed for one of my region?

lanyusea, 2017-01-29:
@AldrichCabrera  yes you can. 

But you may need to modify some javascript files accordingly here: https://github.com/dji-sdk/Onboard-SDK-ROS/tree/3.2/dji_sdk_web_groundstation/webpage_waypoint_client/web_waypoint

AldrichCabrera, 2017-01-30:
@lanyusea Thank you very much. 

It was just changing the coordinates in wp_control.js.

lanyusea, 2017-01-31:
@AldrichCabrera  oh... I thought you mean somthing like changing the Baidu Map into Google Map....

closed 2017-01-31 by lanyusea