dji-sdk/Onboard-SDK-ROS #56: relocation R_X86_64_32 linking failure on dji_sdk_lib

jvgomez, 2016-12-14:
If I try to use the `dji_sdk_lib` on a shared library (a ROS plugin in this case) I get the following error:

```
/usr/bin/ld:~/catkin_ws/devel/lib/libdji_sdk_lib.a(DJI_HotPoint.cpp.o): relocation R_X86_64_32 against `_ZN3DJI10onboardSDK8HotPoint13startCallbackEPNS0_7CoreAPIEPNS0_6HeaderEPv' can not be used when making a shared object; recompile with -fPIC
~/catjub_ws/devel/lib/libdji_sdk_lib.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [~/catkin_ws/devel/lib/libdji_plugin.so] Error 1
make[1]: *** [CMakeFiles/dji_plugin.dir/all] Error 2
make: *** [all] Error 2
```

After searching a bit, it seems that I have to enable the `-fPIC` compiler flag on the `dji_sdk_lib` package:

```
set(CMAKE_CXX_FLAGS "-std=c++11 -fPIC ${CMAKE_CXX_FLAGS}")
```

This works. It seems that it fails if you are trying to compile a shared object in an x64 architecture, in order to include a static library it has to be "Position Idependent Code", which is enabled with `-fPIC` on GCC.

Is there any reason for not doing this? Should I send a PR?

rikba, 2018-03-26:
I have had a similar issue when linking against the SDK from a catkin workspace: 

```
/usr/bin/ld: /usr/local/lib/libdjiosdk-core.a(dji_version.cpp.o): relocation R_X86_64_PC32 against symbol `_ZN3DJI4OSDK7Version2FWEhhhh' can not be used when making a shared object; recompile with -fPIC
```

Added 
```
set(CMAKE_CXX_FLAGS "-fPIC ${CMAKE_CXX_FLAGS}")
```
to [Onboard-SDK/CMakeLists.txt](https://github.com/dji-sdk/Onboard-SDK/blob/3.6/CMakeLists.txt).

vacky11, 2018-10-19:
I had similar error when I was adding a new library to my Cmakelist and DJIOSDK library was linked to this new library.

Instead to changing the Onboard-SDK/Cmakelist, I added my library as a static library and the error was resolved. 
```

add_library(<name> STATIC
            source1 [source2 ...])
```

closed 2021-06-25 by dji-kyle