dji-sdk/Onboard-SDK-ROS #58: How to display the camera to a PC with linux without using Manifold.

AldrichCabrera, 2017-01-11:
Hi everyone, anyone knows how can I visualize the camera **without using** the **Manifold**, or any alternative instead of using the **Manifold.**

andre-nguyen, 2017-01-11:
You can't, unless through some convoluted way you try getting the camera feed from a tablet connected to the remote controller.

dastoHumanitas, 2017-01-12:
My guess would be to use the DJI debug bridge application on a phone connected to the RC and then build either mobile project (android or ios) for a non mobile station.

cheers.

David St-Onge, ing. MGP PhD
Directeur Scientifique Robotique
Solutions Humanitas
humanitas.io [http://humanitas.io]

AldrichCabrera, 2017-01-12:
I also thought about that but it became very complicated. Anyone know where I can get a manifold? The DJI store no longer has it.

Thank you very much to both.

andre-nguyen, 2017-01-12:
You have to contact DJI, probably the enterprise division.

I remember there was a team at the DJI Dev challenge this summer who also wanted to purchase more manifolds and they spoke to Robert Schlub directly.

lanyusea, 2017-01-12:
Manifold is only available for educational developers.
You can contact dev@dji.com for detail.

AldrichCabrera, 2017-01-12:
I'll try to communicate with DJI. Thanks

closed 2017-01-31 by lanyusea