dji-sdk/Onboard-SDK-ROS #60: [A3] SDK version did not match

KahnShi, 2017-02-02:
Hi all,

My drone is S900 with A3 Pro controller.
When I tried to launch sdk_maniford.launch, the error is

> [ INFO] [1486031607.453965129]: Succeed to create thread for readPoll
Broadcast call back received 
ERROR activateCallback,line 648: SDK version did not match
STATUS getDroneVersionCallback,line 595: version ack = 65281
STATUS getDroneVersionCallback,line 604: version name = v1.0 BETA A3-03.02.15.37

I found same problem in latest discussions in DJI Forum, such as [1 ](http://forum.dev.dji.com/thread-33019-1-1.html)and [2](http://forum.dev.dji.com/thread-33028-1-1.html).

All my A3 firmware version is the latest, and Enable API Control is selected. 
I tried to change drone_version in launch file both to "A3_32" and "A3_31", but neither could work.

Is there any advice?
Thank you in advance!

KahnShi, 2017-02-02:
FYI
Current hacky solution:
1. Check your version, for example, mine is v1.0 BETA A3-03.02.15.37
2. [Here ](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/b53e821701011452d507a0f24b555dc9cd2b0df3/dji_sdk_lib/include/dji_sdk_lib/DJI_Version.h#L38) change as Make_VERSION(3, 2, 15, 37))
3. It worked.

Wish maintainer could update and give official support soon.
Thanks!

closed 2017-02-02 by KahnShi

amenonDJI, 2017-02-02:
We will have a fix for this today or tomorrow.

damanfb, 2017-02-02:
Do you know if the fix will work for the following as well:

`STATUS init,line 62: Open serial device /dev/ttyUSB0 with baudrate 230400...`
`[ INFO] [1486070056.769198166]: Succeed to create thread for readPoll`
`Broadcast call back received `
`ERROR activateCallback,line 648: SDK version did not match`
`STATUS getDroneVersionCallback,line 595: version ack = 65281`
`STATUS getDroneVersionCallback,line 604: version name = v1.0 BETA PM820V3PRO-03.02.10.00`