dji-sdk/Onboard-SDK-ROS #61: Matrice M600 Pro: SDK version did not match.

damanfb, 2017-02-03:
I have a Matrice 600 Pro with updated firmware (V1.0.0.80 according to DJI Assistant 2) and I get the following error when I launch the dji_sdk sdk_manifold.launch
'STATUS init,line 62: Open serial device /dev/ttyUSB0 with baudrate 230400...`
`[ INFO] [1486087060.234682586]: Succeed to create thread for readPoll`
`Broadcast call back received `
`ERROR activateCallback,line 648: SDK version did not match`
`STATUS getDroneVersionCallback,line 594: version ack = 65281`
`STATUS getDroneVersionCallback,line 604: version name = v1.0 BETA PM820V3PRO-03.02.10.00`

I set my launch file with the drone_version A3_32 but still can't get it to work.

amenonDJI, 2017-02-03:
@damanfb This is a known issue and we will have a hotfix for it shortly.

damanfb, 2017-02-03:
@amenonDJI I pulled the newest commit for the fix down and get the following output when launching the dji_sdk/sdk_manifold.launch on my Odroid using the M600 Pro

`--- Connection Info ---`
`Serial port: /dev/ttyUSB0`
`Baudrate: 230400`
`-----`
`STATUS init,line 62: Open serial device /dev/ttyUSB0 with baudrate 230400...`
`[ INFO] [1486155935.372025820]: Succeed to create thread for readPoll`
`=================================================`
`Hardware : PM820V3PRO`
`Firmware : 0x03020A00`
`=================================================`
`Broadcast call back received `
`ERROR activateCallback,line 799: SDK version did not match`

damanfb, 2017-02-03:
@amenonDJI @rohitsantdji Is there a problem with it being an A3Pro rather than the base A3?

rohitsantdji, 2017-02-04:
This is a bug in the M600/pro's 1.0.0.80 versioning system; a fix has been issued, please try the latest commit and let us know if the issue has gone away.

damanfb, 2017-02-04:
@rohitsantdji it is working now, thank you very much for the prompt fixes!

closed 2017-02-04 by damanfb