dji-sdk/Onboard-SDK-ROS #63: "New Device, please link DJIGO to your remote controller and try again"

damanfb, 2017-02-14:
I'm not sure why I'm getting this error or how to fix it. As far as I can tell, DJI GO is connected and I can take off and use DJI GO to control my M600 Pro, but I get that error when launching dji_sdk sdk_manifold.launch.

chutsu, 2017-02-19:
did you fill in this line in the launch file?

    <param name="app_id" type="int" value="1234EnterAppIDhere"/>

and 

    <param name="enc_key" type="string" value="1234Enterkeyhere"/>

damanfb, 2017-02-20:
Yes, I filled in those lines. The launch file won't even launch due to an xml error if you do not fill those lines.

amenonDJI, 2017-03-01:
@damanfb I am guessing you have gotten past this issue. Please re-open if you still see this.

damanfb, 2017-10-02:
We are still seeing this with our M600 again and it is preventing a demo from happening with a customer. We have tried power cycling, multiple tablets with DJI Go, etc. What combination of tablet, developer user name, and app id should we be trying?

amenonDJI, 2017-10-02:
@damanfb Could you tell me the DJI Go version that you're using along OSDK and M600 FW version? I will test it out and let you know.

tash-rif, 2017-11-08:
Encountered the same message but I realised that need to set the mode to F on the remote controller.

closed 2021-06-25 by dji-kyle