dji-sdk/Onboard-SDK-ROS #64: How send data to mobile

guicolares, 2017-02-15:
Hi, 

I need send somes datas to mobile using OnboardSDK (Manifold) with python (ROS) and get that data with "flightController:didReceiveDataFromExternalDevice:"

There is not exemples to do that in Python. Has somebody already done that ?

desmosedic, 2017-02-15:
@guicolares We have a C++ implementation of sending data from Onboard SDK to Mobile SDK. Take a look at the ROS sample - Mobile Commands sample, which essentially takes CMD IDs from Mobile, passes it on to Onboard SDK via fromMobileCallback. Sorry, there is no similar python implementation.

amenonDJI, 2017-02-15:
@guicolares We have a C++ implementation of sending data from Onboard SDK to Mobile SDK. Take a look at the ROS sample - Mobile Commands sample, which essentially takes CMD IDs from Mobile, passes it on to Onboard SDK via fromMobileCallback. Sorry, there is no similar python implementation.

guicolares, 2017-02-17:
@amenonDJI Thank you answered my question but how can I execute Python script demo ? With C++ example is working but I want change it.

Thanks again

amenonDJI, 2017-02-17:
@guicolares Sorry, we do not have a python sample for MSDK->OSDK pipeline.

guicolares, 2017-02-17:
@amenonDJI Yes you have, the script is on dji_sdk_demo/script/client.py 
I just want execute it. 

Thanks

guicolares, 2017-02-17:
@amenonDJI How can I send data from OnBoard SDK to mobile SDK in C++ implementation demo ? 

Thanks.

amenonDJI, 2017-02-17:
@guicolares  The python client sample has not been updated or supported for the last few releases and there is no immediate plans to support it. 

The current ROS implementation only supports MSDK->OSDK and not the other way round. You can send data from OSDK -> MSDK via the Linux sample. You can read about this implementation [here](http://developer.dji.com/onboard-sdk/documentation/github-platform-docs/Linux/README.html#modes-of-operation). Please read all about MSDK <-> OSDK communication [here](http://developer.dji.com/onboard-sdk/documentation/github-platform-docs/MobileOnboardSDK/Mobile-OSDK.html).

amenonDJI, 2017-02-22:
@guicolares Feel free to open a new issue if you have any further questions.

closed 2017-02-22 by amenonDJI

lanyusea, 2017-03-11:
@guicolares    see if you can try this #74 
should work but I have no drone to test