dji-sdk/Onboard-SDK-ROS #65: M600 not holding position

damanfb, 2017-02-17:
I have an M600 using the ROS connection of the Onboard SDK. I have a node that works in the sim to hover at a desired height in the local position, but on the real drone it immediately loses altitude quickly until the pilot recovers in P-Mode.

I have logged the messages and can confirm when it's in F-mode, the flight_control_info message says the control mode is 12 (horiz pos, vert pos, and yaw rate), the cur_ctrl_dev_in_navi mode is 2 (onboard) and serial_req_status is 1 which all match the sim. The local position message shows the correct heights we were flying at, and the desired height it set at 10m. I'm using the DJIDrone method attitude_control(...)

amenonDJI, 2017-02-17:
@damanfb Have you had a chance to try the Attitude control sample in ROS and is your node based off of this sample? Could you give us more information on the exact commands and the frequency at which it is being sent from the OES to the Flight Controller?

damanfb, 2017-02-17:
@amenonDJI I did look at the ROS sample in the dji_sdk_demo. I pulled from that for the most part.

I have my node running at 50Hz on an Odroid XU4 using an Adafruit TTY-USB converter to connect to the API port on the A3. The node checks for the F-mode switch and to be in flight status=2 (flying) and requests control using the dji_sdk::PermissionControlClient service and I get a true result from that, both in sim and experiments. 

For the control message I use the following 
`attitudeControlSrv.request.flag =`
`	  DJI::onboardSDK::Flight::HorizontalLogic::HORIZONTAL_POSITION |`
`	  DJI::onboardSDK::Flight::VerticalLogic::VERTICAL_POSITION |`
`	  DJI::onboardSDK::Flight::YawLogic::YAW_RATE |`
`	  DJI::onboardSDK::Flight::HorizontalCoordinate::HORIZONTAL_GROUND |`
`	  DJI::onboardSDK::Flight::SmoothMode::SMOOTH_ENABLE;`
`	x = 0.0;`
`	y= 0.0;`
`	z = nominalAltitude;`
`	yaw = 120.0 * rcChannels.yaw / 10000.0;`
`djiDrone->attitude_control(attitudeFlag, x, y, z, yaw);'
where rcChannels is a message from /dji_sdk/rc_channels, and djiDrone is initialized as
`DJIDrone* djiDrone; djiDrone = new DJIDrone(nh);` where nh is my node handle.

I have confirmed in both sim and experiments the flight control mode is 12, matching horiz_pos_vert_pos_yaw_rate as expected, I get onboard control listed, and in the sim it will hover at nominalAltitude (which is read in from a rosparam). 

I have checked the logs and the GPS health status in /dji_sdk/global_position is 4-5 the entire time.

damanfb, 2017-02-17:
I did a rosbag record -a as well and have the bags and their csv's if it will help...

amenonDJI, 2017-02-17:
Yes, the rosbag will help. Also, just to confirm, are you sending out 0x99 as the control flag?

damanfb, 2017-02-17:
The bag is too large of a file at the moment. I'll see if I can split it. In the meantime, for the control flag, do you mean the first term in the method attitude_control(). I  have set it using the typedefs that I provided in the previous response. I believe those come out to 0x99. From DJI_Flight.h in dji_sdk_lib has vert_pos = 0x01, hori_pos = 0x80, yaw_rate = 0.08, and smooth=0x01. If I did my math right that's 0x99.

damanfb, 2017-02-17:
Here are the bags.
[dji_bags.zip](https://github.com/dji-sdk/Onboard-SDK-ROS/files/782234/dji_bags.zip)

amenonDJI, 2017-02-17:
@damanfb Yes, your control mode looks right. I have been looking at your logs and it does look like the aircraft looses altitude after the /drone_have_control changes to True. From your code -  is your Parameter server for NominalAltitude being set somewhere else in your project? Otherwise, the default altitude is 10, which is good.   

In order to test this, I'd like to know your M600 A3 firmware version and Onboard SDK version. Thanks.

damanfb, 2017-02-17:
@amenonDJI no, the altitude parameter is not being set elsewhere. I output the values of x, y, and z to the message /drone_control_vector and it shows as 10 when I checked the log. We have also tested varying numbers between 5-100 and get the same result. We tested the attitude control demo from the dji_sdk_demo and it did work so it seems the velocity control is working (says control mode 3 when I watch the flight_control_info message) but not position control.

I'll edit and add the firmware as soon as I get it. I have pulled the onboard sdk after the last push you made for the version info fix. I'll add the screenshot of the "roslaunch dji_sdk sdk_manifold" as soon as I can.

amenonDJI, 2017-02-22:
@damanfb We have not had a chance to test the M600 with your test conditions due to bad weather, hoping to test tomorrow.

damanfb, 2017-02-22:
@amenonDJI Thanks for the update.. We have been grounded for weather this week as well. I made a test node that traces out a square and can set the mode at launch to iterate over every combination of position and velocity control in horizontal, vertical, and yaw. This will hopefully confirm which modes are working since the attitude control demo which uses control mode 0x40 worked.

damanfb, 2017-02-23:
@amenonDJI Just an update, we had a break in weather today and we flew my new test node I made. I had one where we could go through each combination of position or velocity for horizontal, vertical, and yaw. The vertical position combined with any other control mode had the same falling behavior. Vertical velocity worked with every other control mode. It seems the issue is isolated with the vertical position in our testing.

amenonDJI, 2017-02-25:
@damanfb Thanks for the update. We got a chance to test this out yesterday and noticed inconsistent behavior as well. I am talking to an internal team to confirm the bug. Keep you posted. Meanwhile, could you send me your FC firmware version - you can find the firmware version in your "Firmware update" menu item on the left in DJI Assistant.

damanfb, 2017-02-27:
@amenonDJI 

The A3 firmware version is V1.0.0.80, the Lightbridge version is V1.1.50, and the DJI Assistant version is V1.0.7.

amenonDJI, 2017-03-01:
@damanfb We can confirm that this is a bug that was introduced into the Flight Controller. The reason this is happening is that the VERT_POS mode handles the throttle input as absolute altitude, not the relative height to home point anymore. In simulator the airplane is set to be at the altitude of 0, so this case didn’t arise. For the emergency solution, you can add the relative height to the airplane take off altitude. This result will be added to your VERTICAL_POS commands.

We apologize for this issue and we are looking to resolve it as soon as possible.

andre-nguyen, 2017-03-02:
@amenonDJI Is there a possibility that this also affects the M100?

amenonDJI, 2017-03-02:
@andre-nguyen We tested the M100 recently. It does not affect the M100.

amenonDJI, 2017-03-09:
@damanfb Did the work around fix your issue of M600 losing altitude? For your information, we are releasing a hotfix soon that will handle this issue in the Core library.

damanfb, 2017-03-09:
@amenon DJI I never implemented that workaround because I had already implemented my own PID controller for holding position with velocity as the controller output. Then when you release a fix I can just delete the line calling the vertical PID.

amenonDJI, 2017-03-09:
@damanfb Great, that sounds good.

closed 2017-03-09 by amenonDJI

amenonDJI, 2017-03-15:
@damanfb  A [hotfix](https://github.com/dji-sdk/Onboard-SDK-ROS/releases/tag/3.2.2) has been released for the Attitude Control Vertical position bug. Please pull the latest code to get the fixes.