dji-sdk/Onboard-SDK-ROS #66: local position outputted by the is not NED

chutsu, 2017-02-19:
According to the mesage definition [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk/msg/LocalPosition.msg), the local position is supposed to be in NED, but it is in NEU instead.

The bug can be found at [here](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk/src/modules/dji_sdk_node_main.cpp#L94).

andre-nguyen, 2017-02-19:
It's not a bug it's a "feature" for non-academic people or people with no formal training in navigation systems. It's unfortunate because it messed with my team quite a bit.

Here's the bit in the documentation that will interest you https://developer.dji.com/onboard-sdk/documentation/introduction/things-to-know.html#2-ground-frame

See the note:

> Note: To alleviate this issue of the unnatural downward-pointing positive Z, we adjust the direction of vertical control in order to make the height or vertical velocity to be positive upwards. In other words, giving a positive velocity will make the UAV ascend. This adjustment does not affect the directions and the orders of the other two axis. We are not defining a left-handed co-ordinate system; the actual computations on the flight controller happen in an NED frame.

Also you can see in the message definition that `-Z` is down, so the message correctly indicates that `+Z` is up.

chutsu, 2017-02-19:
Thanks, I've been bitten by co-ordinate transforms with MAVROS as well... Good job on your High Speed landing paper btw :P

closed 2017-03-01 by amenonDJI