dji-sdk/Onboard-SDK-ROS #67: flight status only showing 0,1,2

damanfb, 2017-02-21:
I see in the documentation the flight status should show 1 - standby, 2 - taking off, 3 - in air, 4 -landing, 5 - finish_landing. However, when I look at the bags from a flight and the live data from a sim the status shows up as 0 until I take off, it shows 1 when I take off, then 2 for the remainder of the time in the air and when landing and switches back to 0.

I also see in DJI_Flight.h [#83](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk_lib/include/dji_sdk_lib/DJI_Flight.h#L83) the same for 1 -standby, 2 - takeoff, 3 - in_air, 4 - landing, 5 - finish_landing

amenonDJI, 2017-03-01:
@damanfb Sorry for the late reply, I'll test this out tomorrow and get back to you shortly.

rohitsantdji, 2017-03-06:
@damanfb The flight status enums have been simplified since the latest M600/A3 FW's to 0: Motor stopped; 1: Motor started, aircraft on ground; 2: In air. We apologize for the lack of documentation on this. We will push out a documentation update soon.

closed 2017-03-06 by damanfb