dji-sdk/Onboard-SDK-ROS #68: ERROR activate,line 388: Please call getDroneVersion first.

YorksonChang, 2017-02-26:
When I tried to launch the DJI ROS core package it info me an error:

ubuntu@tegra-ubuntu:~$ roslaunch dji_sdk sdk_manifold.launch
... logging to /home/ubuntu/.ros/log/94b0d0d2-fc32-11e6-979d-60601f1999cd/roslaunch-tegra-ubuntu-15212.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:39351/

SUMMARY
========

PARAMETERS
 * /dji_sdk/app_bundle_id: Not required in t...
 * /dji_sdk/app_id: 1034014
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: ad3d034e49f21dd24...
 * /dji_sdk/groundstation_enable: 1
 * /dji_sdk/serial_name: /dev/ttyTHS1
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[dji_sdk-1]: started with pid [15230]
=================================================
app id   : 1034014
app key  : ad3d034e49f21dd24d75c4885febfcf19c54745c19ef4968ceefa3bd7d0ea0dd
=================================================
--- Connection Info ---
Serial port: /dev/ttyTHS1
Baudrate: 230400
-----
STATUS init,line 62: Open serial device /dev/ttyTHS1 with baudrate 230400...
[ INFO] [1488121163.298959031]: Succeed to create thread for readPoll
=================================================
Hardware :
Firmware : 0x00
=================================================
**ERROR activate,line 388: Please call getDroneVersion first.**
Broadcast call back received

Can anyone tell me what's wrong with my manifold ???
Thanks guys

amenonDJI, 2017-03-01:
@YorksonChang Is this is the first time you're connecting to the drone with the DJI ROS package? If not, when did you start seeing this issue? 
- Have you enabled API on DJI Assistant?
- Do you have the correct baud rate set on Assistant and on the ROS package? 
- What aircraft are you using? 
- Do you have the Tx and Rx cables connected to the right pins?
- Do you have the latest code from the repo?

closed 2017-03-03 by YorksonChang

6gekang, 2017-03-09:
how did you solve this problem

abdullah38rcc, 2017-03-09:
Hello everyone. I have the same problem. I am using Jetson Tegra board tk1. It is running 14.04 ubuntu. I made a successful connection using Ubuntu 16.04. But when i have tried to connect with my Jetson board, it showed broadcast call received. Didn't activate.

Now, after upgrading firmware to the last version, It is showing getDrone version error !! Tried everything and still it is showing the same error !! Is it happening for update error ? Please anyone answer it. I need urgent solution. Now feeling so hopeless. Any kind of help is appreciated. I tried command line even. With this new update nothing is working !! Please someone look int it or DJI developers please check it.

YorksonChang, 2017-03-12:
I just try another manifold...I think there is something wrong with my serial port of my box(Hardware issue).

abdullah38rcc, 2017-03-20:
@YorksonChang  Could you please explain which manifold is working for you ? Unfortunately i have the same problem and couldn't get any good solution. Is it happening for the new update or i need to try with new laptop or manifold to overcome it. 

@amenonDJI  Hi Arjun, really i will appreciate if you can explain do i need to check my firmware update problem or not ? I didn't know why it is happening. I think many of us having this problem and I have tried what you have asked to do. But didn't work. Please help me in this regard.