dji-sdk/Onboard-SDK-ROS #69: DJI-SDK Demo fails and drone falls from the sky

lvale, 2017-02-28:
Our setup is using a DJI A3 1.7.0 connected to a Odroid running the DJI-SDK demo v3.2.1. We also use a LB2 to control the system.

We've encountered a problem with our code, and to be sure we've used the demo code and the problem is exactly the same.

Using the simulator no issue is found and everything and everything works as expected, but on the field the drone just falls from the sky. We've checked the logs and the motors seem to go to idle. Apparently we' still have a bit of yaw control while falling, but it's irrelevant because the drone is falling :)

We didn't manage to check if X and Y are still working.

We've tried Local Position Control and Global Position Control and the Circle demo (after the request of the SDK permission was completed with success) and the drone falls.. So, no Z control.

We then tried using altitude instead of height in the Local Position Control and the results were still a falling drone....

So the demo code fails on a real vehicle but is ok on the sim. 

We've opened an issue also on DJI dev forum.

damanfb, 2017-03-01:
I have been also encountering this and have an issue at [#65](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/65). I was able to find that the issue seems to be isolated to the Vertical Position control in my case. When using vertical velocity control and I implemented my own PID for position, I have been successful.

lvale, 2017-03-01:
Thanks, We'll try to replicate your workaround here.

@amenonDJI could you please give us an update for the final solution (I believe the issue resides on the A3)

amenonDJI, 2017-03-01:
@lvale We are analyzing logs and will get back to you shortly.

amenonDJI, 2017-03-01:
@lvale We can confirm that this is a bug that was introduced into the Flight Controller. The reason this is happening is that the VERT_POS mode handles the throttle input as absolute altitude, not the relative height to home point anymore.  In simulator the airplane is set to be at the altitude of 0, so this case didn’t arise. For the emergency solution, you can add the relative height to the airplane take off altitude.  This result will be added to your VERTICAL_POS commands. 

We apologize for this issue and we are looking to resolve it as soon as possible.

lvale, 2017-03-01:
@amenonDJI Thanks. I know this is a tough question, but do you have a timeframe to publishing the corrected A3 firmware, even if on a "pre-release" version ?

lvale, 2017-03-01:
@amenonDJI We've just tested with your advice of VERTICAL_POS and still doesn't work.

We used Local Position, we tried with the altitude from the Global Position plus 10 meters and also with the altitude from the A3 GPS topic plus 10m and it still fell of the sky. 

Using attitude control with Horizontal position and Vertical velocity works. @antoniocoratelli will add the rosbags

antoniocoratelli, 2017-03-01:
These are the rosbags of the test flights we just did. 

The first one is using _AttitudeControl_ using a simple P Controllers for XY position offset and Z velocity; we managed to control the drone to go where we wanted.

In the second bag we tested the _LocalPositionControl_ trying to go 10 meters above ground: ground was measured to be 160m above sea level using the information from `/dji_sdk/A3_GPS` topic, thus we sent a _LocalPositionControl_ message with 170m. The result was still a falling drone.

Both rosbags are [here](https://github.com/dji-sdk/Onboard-SDK-ROS/files/811056/dji_demo_test.zip).

amenonDJI, 2017-03-01:
@lvale Could you please send us the FC log files as well? In order to do this:
- Go the menu tab (**...** icon on the right top corner) on DJI Go. 
- Click on the quadcopter icon and go the Advanced Settings menu 
- At the end of this menu you will see the button to Enter Flight Data Mode. Enter this mode. 
- When you connect the aircraft to the computer, it will show up as a storage device with the logs in it. 
- Zip the log files and send them over to us

antoniocoratelli, 2017-03-02:
@amenonDJI I have a zip file with the logs, since it's quite big (~700 MB) can you give me your email address so I can send you a wetransfer?

amenonDJI, 2017-03-02:
@antoniocoratelli You can use the email arjun.menon@dji.com. As an option, you could also use Dropbox.

antoniocoratelli, 2017-03-02:
@amenonDJI here's the link https://we.tl/p6J7FO7wOu

lvale, 2017-03-06:
@amenonDJI Anything else you need from us ?

amenonDJI, 2017-03-07:
@lvale @antoniocoratelli We did some testing today on our drone to make sure that the solution worked. We can confirm that the workaround to add Altitude to relative height works. Few questions for you:

- Would you be able to upgrade to the latest firmware. This will ensure consistency between our tests. 
- I noticed in your bag file that the Altitude reported is 70m. You mentioned you saw 160m. Could you try adding the global_position altitude (instead of A3 GPS altitude) to the relative height? 
- Altitude can vary quite a bit, I am assuming that the altitude is being added in code just before the call to the attitude control function, instead of adding the Altitude value that you visually inspected.

antoniocoratelli, 2017-03-07:
Hi @amenonDJI ,
the firmware we have on our A3 is v1.7.0, is it the latest one?

After the initial (failing) test with our code we only used the `client.cpp` in the `dji_sdk_demo` package, adding just an `std::cin` in [case 14](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk_demo/src/client.cpp#L440) in order to get altitude from keyboard.

Although we recorded just the bag with the 160m offset, we did other tests adding 70m and other altitudes for `local_position_control`, but the results were exactly the same.

thanks for your time,
Antonio

amenonDJI, 2017-03-07:
@antoniocoratelli @lvale I think I see what the problem is, by local position control, do you mean the **local_position_navigation_send_request** API?  Could you please use attitude_control API and use the Z_position control flag instead? The local_position_navigation_send_request API may have additional bugs we need to investigate. 

Here is an example of the API being used: 
drone->attitude_control(0x99, 0, 0, 10, 0);
where 0x99 =   0x01 (vert_pos) + 0x80 (hori_pos) + 0x08 (yaw_rate) + 0x01 (smooth_enable) 

If you would like to read more about the specific control modes, please take a look at [this](https://developer.dji.com/onboard-sdk/documentation/appendix/index.html#control-mode-byte) document.

antoniocoratelli, 2017-03-08:
@amenonDJI  what we did at the end was using `attitude_control` with flag `0b10000001` with 3 simple P controllers for X Y and Z axis like in the following scheme, so we have a workaround, but controlling directly the drone with `local_position` would be better

```
axis setpoint      error         velocity command              axis value
  ------> + (   )  ------> | P | ----------------> | DJI SDK | --------->
                                                                     |
            - ^----------------------------------------------------- |

```

amenonDJI, 2017-03-08:
@antoniocoratelli You can use the **attitude_control** API to control the aircraft using **local Z position**. For this, please use the API **drone->attitude_control(Control Flag, 0, 0, 10 + altitude, 0);** where **Control Flag** is **0b10010001** and **altitude** is the **take off altitude** read from the **/dji_sdk/global_position topic**.  

Please do not use the **local_position_navigation_send_request** since this API requires further investigation for additional bugs.

amenonDJI, 2017-03-15:
@antoniocoratelli @lvale A [hotfix](https://github.com/dji-sdk/Onboard-SDK-ROS/releases/tag/3.2.2) has been released for the Attitude Control Vertical position bug. Please pull the latest code to get the fixes.

lvale, 2017-03-16:
Thanks @amenonDJI 
We'll get feedback after testing

tariq-r, 2017-03-17:
We pulled the hotfix and tested today using the Linux interactive sample,  Draw Sqaure.  The sample was initiated at 3 meters above the ground.  The aircraft immediately yawed due north and began climbing.  The climb was stopped after about +15 meters by switching to attitude mode on the remote controller.  Flight controller is A3,  Firmware version is 1.7.   It appears that for the draw square demo, the problem remains.   

We also used the SDK to build a simple, move right, move left, UI,  in this case also, when a movement command is sent, the aircraft starts climbing.

We will attempt to use attitufude_control as work around so that we can continue our development program but the OSDK is not functioning properly in flight.    Are you able to reproduce this bug with an aircraft (not the simulator)

amenonDJI, 2017-03-17:
@tariq-r 

"We pulled the hotfix and tested today using the Linux interactive sample, Draw Sqaure. The sample was initiated at 3 meters above the ground. The aircraft immediately yawed due north and began climbing. The climb was stopped after about +15 meters by switching to attitude mode on the remote controller."

Almost expected behavior, the Draw a Square sample is directed to go to 10 meters from where it was initiated. Ideally, it should've stopped at 13 or 14 meters. 

"We also used the SDK to build a simple, move right, move left, UI, in this case also, when a movement command is sent, the aircraft starts climbing."

What command did you use for this? Could you please copy the commands here. Also, how long did it climb for? Did it stop climbing at 5m? 

Our fix was tested on a real aircraft and was confirmed to be working. The fix is very simple, if it sees the aircraft taking off, it sets the homepoint altitude. It then adds the homepoint altitude to the commanded Z if the firmware is in the list of versions affected by this bug.

tariq-r, 2017-03-20:
Hello

For the square we of course are using from LinuxFlight.cpp

```
int drawSqrPosCtrlSample(CoreAPI* api, Flight* flight) 
```
which uses
```
int positionControlStatus = moveByPositionOffset(api, flight, 0, 0, 10, 0);
```

In our code we are using 

```
int r = moveByPositionOffset(api, flight, x, y, z, yaw);
```

Should it make a difference if we are using sbus and not lightbridge 2?

Also we noticed that the automatic takeoff does not work,  the sdk sample accepts the command but we have to throttle up on the sbus radio to do the takeoff.  Could this be part of the problem?

We are going to try the SDK with lightbridge 2 today.

Thank you for the feedback, hope to continue to the dialog until we get this working.  For your knowledge, our application involves using distance sensors to fly close to surfaces, so we need relatively fine control.  Is moveByPositionOffset the best command to use for this?  We will be sampling distance at ~ 200 Hz, to maintain a distance within 10 cm from a surface.

Thank you
Tariq

tariq-r, 2017-03-20:
@amenonDJI  we just tried the linux sample with the lightbridge 2,  the SDK arms the aircraft properly (A3 quadcopter), when we send the automatic takeoff via the SDK sample, the interactive sample indicates that the takeoff has started but the takeoff does not start.  we have to do this with the lightbridge 2 remote controller.  

But good news,  the Draw Square sample works perfectly!     We are going to try with our own software shortly.  Any idea why the DrawSquare from the sample would work with Lightbridge 2 and not with SBUS?

amenonDJI, 2017-03-20:
@tariq-r The takeoff command only works if the propellers are disarmed. This is expected behavior. Good to know that the DrawSquare works with LB2. I will raise a bug on your behalf for the s-bus. Could you also send me any flight logs you have?

tariq-r, 2017-03-21:
@amenonDJI  I understand about the takeoff command, thank you.   Here is a folder containing flight logs both with sbus and with lb2.  https://www.dropbox.com/sh/dn32u42ocmotb2w/AAC8JY3ww0sDW0K5YHtV4Igda?dl=0

Thanks again for your help.  We will continue to test our code to make sure that we can replicate the sample.

amenonDJI, 2017-03-24:
@tariq-r We tested with S-bus (14 channel Futaba receiver with T14SG Transmitter) running the latest A3 firmware (1.7.0) and we were not able to reproduce the issue. The logs don't reveal the issue either. I would recommend testing with smaller Z inputs, maybe 5m and see if the S-bus issue continues. 

We did see issues with the TakeOff command using S-bus . I will do some more tests to confirm behavior and raise a bug.

tariq-r, 2017-03-25:
Ok, tthank you @amenonDJI  we will do some more tests as well trying smaller Z inputs

amenonDJI, 2017-03-30:
@tariq-r If you continue to see the problem, please open a new Github issue. Closing this.

closed 2017-03-30 by amenonDJI

tariq-r, 2017-03-31:
@amenonDJI   Thank you, we are able to execute the linux square sample with sbus properly in the simulator and with the aircraft.