dji-sdk/Onboard-SDK-ROS #72: dji sdk node constantly crashing

damanfb, 2017-03-09:
While running with the demo node or custom control nodes, the dji skd node continually crashes every few minutes. We are launching it using "roslaunch dji_sdk sdk_manifold.launch." We have the the following stack trace from the crash:

Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00000000 in ?? ()
[Current thread is 1 (Thread 0xafeff420 (LWP 10004))]
(gdb) info stack
#0  0x00000000 in ?? ()
#1  0x00366440 in DJI::onboardSDK::CoreAPI::readPoll() ()
#2  0x0029a5fa in DJI::onboardSDK::ROSAdapter::APIRecvThread(void*) ()
#3  0xb6caf5b4 in start_thread (arg=0x0) at pthread_create.c:335
#4  0xb6aa2aac in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:89
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

We are using a M600 Pro with A3 firmware version is V1.0.0.80, the Lightbridge version is V1.1.50, and the DJI Assistant version is V1.0.7.

amenonDJI, 2017-03-09:
@damanfb Could you also confirm your OES platform? Is it a TK1 running Ubuntu 14.04?

damanfb, 2017-03-09:
Odroid XU4 running Ubuntu Mate 16.04

closed 2017-03-30 by amenonDJI