dji-sdk/Onboard-SDK-ROS #73: ERROR activate, line 388 : call getDroneVersion first

abdullah38rcc, 2017-03-11:
I'm using an M100. It used to work before the last update. I have been developing our code for an algorithm test. Firstly, the issue was same for us. It worked fine with same pc having windows 10 and ubuntu 16.04 with ROS kinetic installed. USB port was '/dev/ttyUSB0' for them. But when i have used Nvidia Jetson tk1 board for our project it has Ubuntu 14.04 installed with ROS indigo, it showed us following line: Stuck in 'Broadcast callback received' here is a detail issue:

http://forum.dev.dji.com/thread-32671-1-1.html

Secondly, we tried to solve the USB port issue with setting up port '/dev/ttyTHS1'. That time is the February month. After that we updated with latest version of OSDK. Now, ROS it is showing another problem: ERROR activate,line 388: Please call getDroneVersion first

read detail issue :

https://github.com/dji-sdk/Onboard-SDK-ROS/issues/68

We are having this activation problem now !! Even my pc which worked before is now showing same error with the new update. I have tried with 3.1 and 2.3 OSDK also, nothing worked. Ending up with same error. In addition, we do not have the camera, so during new update we faced some problem but finally updated the drone version.

Please suggest me a solution. We are preparing for a research project submission and near deadline having such issues unsolved. We need your serious help to overcome it.

ahmetk6135, 2017-03-11:
I have same problem with abdullah38rcc. I also need urgently solution about that topic. I think upgrade caused this error.Can anybody help.

lanyusea, 2017-03-11:
@ahmetk6135 @abdullah38rcc let's check the logic together:

To active the drone, you need to do select the drone's version first.
To simplify, this part is done in our code, we do the version inquiry first, by the result of which, we active.

The code is here: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk/src/modules/dji_sdk_node_main.cpp#L354

->we call the getDroneVersion(); 
->wait 1 sec for the result
->printf the result
->do the activation, which needs the value stored by getDroneVersion();

in #68 the printf is empty
```
Hardware :
Firmware : 0x00
```

which means the code failed to reach the drone. 
Also, it is the same reason why it is stuck in 'Broadcast callback received', that your code cannot receive data from the drone.

Therefore, I believe, this is a hardware issue or a config issue, instead of bugs in our code.
I suggest you to check all these points @amenonDJI  mentioned in #68 

```
Have you enabled API on DJI Assistant?
Do you have the correct baud rate set on Assistant and on the ROS package?
What aircraft are you using?
Do you have the Tx and Rx cables connected to the right pins?
Do you have the latest code from the repo?
```

abdullah38rcc, 2017-03-20:
@lanyusea  Yes, I am agreed what you have shown us as logic. But believe me we tried everything you have asked to do but didn't work. Did you try your self with without laptop ? What we are asking is : 

I have a jetson tk1 , running ros on ubuntu 14.04, connected to the flight controller using a USD to UART cable. I have modified the sdk_manifold.launch file with the app id and key that I received from DJI. I have set the serial name to '/dev/ttyUSB0' which was the port that the usb was connected to on my computer. I tried using a usb hub and also connecting directly to the TK1's usb port. When I run sdk_manifold.launch ,it says succeed to create thread,Broadcast call back received and just stays there. It does not get activated. My phone, on which I have logged into my dji account, is connected to the remote controller. The RC is also in F mode. I can activate it when I connect it to my pc running ubuntu 14.04. I am at a loss here. 

You have shown the same reason that it will work but this issue was not answered by the users that it worked. So, without solving this issue when new hot fix was updated and new problem arrived with get drone version first"

As we have 2 weeks to submit our research project now I am really in trouble. I would request you if you have a chance to try it in Jetson Tegra tk board. If it works then great.

I have explained that my first setup with 16.04 ubuntu and laptop worked fine. Jetson board never had a connection until now. Currently after new firmware update it lost connection with my laptop installed 16.04. What do you suggest me now ? Should i manually downgrade my firmware version ?

amenonDJI, 2017-03-21:
@abdullah38rcc We just tested this on a Jetson TK1 running Ubuntu 14.04 and an M100 with the latest OSDK code connected to THS1 port and did not have any issues running the sample. 

1. Do not use USB to Serial cable on TK1. Only use the ttyTHS1 port.

2. After upgrading firmware, did you enable API? Every firmware upgrade requires re-enabling API. 

3. Please give us screenshots of your Assistant - SDK page 

4. Please give us screenshots of the ROS terminal output.

lanyusea, 2017-03-21:
@abdullah38rcc 

could you `cat` the `/dev/ttyUSB0` after connected to the drone. check if there is any data print?

abdullah38rcc, 2017-03-21:
@lanyusea  Yes, I will. Last day it was sending random data at our COM4 port. That means the USB cable works but not doing  it correctly. I will post cat of /dev/ttyUSB0 here.

abdullah38rcc, 2017-03-21:
@amenonDJI Thanks for your concern indeed. I am posting all of your requirements here to check.

abdullah38rcc, 2017-03-22:
@amenonDJI  I think i have a problem to connect with ttyTHS1 port. The board has a single USB port and mini USB extent. So I have connected DJI's  UART usb  to the single USB port. But I think no data connection was established. Please show me how to make that. I am sharing all my screenshots. Help needed ! 

[image]

[image]

[image]
[image]
[image]
[image]

abdullah38rcc, 2017-03-22:
@lanyusea Only good news is it  worked for my laptop again. Firstly i have downgraded my drone and then again updated it to the latest. Both of them worked for cat the /dev/ttyUSB0 port. I can see some garbage data on screen. But when i have tried it for cat the /dev/ttyTHS1 for my jetson TK1 board nothing showed on screen !! Please help me to solve this port problem.

rohitsantdji, 2017-03-22:
@abdullah38rcc It seems you are not connecting to your /dev/ttyTHS1 port at all. Connecting DJI UART to the single USB port will try to establish a connection to /dev/ttyUSB0.

Two of the /dev/ttyTHSx UART ports on a TK1 dev kit are mapped to pins on the GPIO expansion bus J3A2. One is mapped to the DB9 connector. Please see these links from NVIDIA's forum to figure out which pins on the J3A2/DB9 to connect to the DJI UART. 

1. http://developer.download.nvidia.com/embedded/jetson/TK1/2014-03-24/JetsonTK1_ModuleSpecification_PM375_V1.01.pdf
2. https://devtalk.nvidia.com/default/topic/820092/set-up-usual-uart/
3. http://www.jetsonhacks.com/2015/09/03/level-shifting-uart-and-gps-part-1-nvidia-jetson-tk1/

Note that you will end up with just wire connections going from the drone to the TK1, no USB connection will be involved. Also be careful with the voltage levels on the TK1 (default is 1.8V TTL i believe, ours is a 3.3v logic level). If you have any further questions about the TK1 specifically, I would recommend checking out the NVIDIA forums.

amenonDJI, 2017-03-30:
@abdullah38rcc Have you tried the ttyTHSx connection? Please let us know your test results.

abdullah38rcc, 2017-04-02:
@rohitsantdji Hi Rohit. I have tried but didn't work.  You have said that the  voltage levels on the TK1 (default is 1.8V TTL i believe, ours is a 3.3v logic level).  Yes, it is and i need a solution for it. Do you have the details to make it work ? Can you suggest any detail solution or screenshots regarding this problem ? 

@amenonDJI Yes, but no success. I need to solve it for my case. As Rohit explained it that i need to check  the voltage level. Please help !

rohitsantdji, 2017-04-20:
@abdullah38rcc - the links I provided should have instructions on how to fix the voltage levels. Since this issue is quite outside the purview of DJI OSDK, I would recommend consulting NVIDIA docs/forums for further assistance.

closed 2017-04-20 by rohitsantdji

horstrand, 2017-09-05:
Hi,

I'm also working with the Jetson TK1, trying to connect to the API port on the Matrice 600.
I'm using the ttyTHS1 port on the Jetson with 115200 Baudrate on both sides (the UserConfig.txt file and on the DJI Assistant). The API port is also activated on the DJI Assitant.

When I run the mobile_sample I get the following activation error:

file:///home/ubuntu/Pictures/OSDK_failed_activation.png

I have upgraded yesterday the firmware of the Matrice 600 to the version 1.0.1.60 (Flight Controller version 3.2.41.5). And communication seems to be working, since it manages to get the firmware version of the drone.
I have also set the app_id and the app_key on the UserConfig.txt

Can you tell me what could be going wrong in the whole setup?

horstrand, 2017-09-05:
[image]