dji-sdk/Onboard-SDK-ROS #75: Cannot manual override using P mode with Matrice 600.

sf624, 2017-03-15:
I cannot manual override the SDK's control using P mode. With matrice 100 I could override by P and A mode, but with the Matrice 600 it seems the override only working on A mode.

I've seen a release note "Notes for using Onboard SDK with the new A3 v1.5.0.0 FW" that says
* If you desire a manual override (a switch position that disables the Onboard SDK), you can go to DJI Assistant 2 ->Basic Settings -> Remote Controller Settings and check the box that says Multiple Flight Modes.

but I could not find such "Remote Controller Settings" in DJI Assistant 2. Also this release note says the mode switch was "replaced by (P,S,A) respectively", but my controller keeps the original (P,A,F) functions according to the DJI GO display. Or is this release note only for naked A3, not A3 attached with M600?

Any help? I'm using up-to-date OSDK, M600 firmware and DJI Assistant 2.

Software versions
OSDK: 3.2.2
M600 firmware: V1.0.0.80
Lightbridge firmware: V1.1.50
DJI Assistant 2: v1.0.9

[image]

rohitsantdji, 2017-03-15:
This is (mostly) expected behavior, with one important caveat.

1. As you correctly guessed, the "Enable Multiple Flight Modes" in the release note is only for standalone A3/N3 systems. As of FW 1.0.0.80 for M600, the "Remote Controller" tab is not exposed since it is meant to work only with the integrated LB2. 

2. Regarding the release note's comments on multiple flight modes - The M600's behavior is consistent with the release note's expectations. Multiple flight modes is implicitly disabled, so the SDK works in all (position-based) modes, since all (position-based) modes are mapped to P mode.

There is an important feature in the M600 behavior you see - you can **manually take over control by simply toggling the switch positions** from P --> F or from F --> P.

The design goal for this behavior is mainly to remove the confusion regarding modes. Put simply, the behavior lets you obtain OSDK control regardless of mode and lets you manually takeover regardless of mode, by toggling the switch. Note that A mode is excluded from these discussions since it is not a position-based mode.

The main point to note is that this takeover is now based on toggling, instead of the resulting mode. For example, If you were doing SDK control in F mode and toggled to P mode, you would successfully take over manually. Then, leaving the mode switch in P mode, you would be able to send an `obtain control` command via the OSDK and successfully obtain control (This sort of behavior would not be possible previously, when the SDK demanded that you be in F mode to obtain control). Finally, after using the SDK to control the craft while your switch is in P mode, if you toggled the switch to F mode again, you would be able to takeover manually too.

We would greatly appreciate it if you could test this behavior and update this post.

sf624, 2017-03-15:
*  you can manually take over control by simply toggling the switch positions from P --> F or from F --> P

Is this behavior described in any articles already published? This is a critical point to know, and I think this implementation is rather making it more confusing and dangerous. 

I've previously made a program to ``` obtain control ``` whenever it detects the drone is not in API mode. Doing so, in matrice 100, I was able to safely manual control in "P" (it ignored the ``` obtain control ``` command) and also immediately revert to API mode in "F". But in Matrice 600, according to your explanation, if I use the same program, when I toggle the switch position it might manual override just for a moment, but the program will immediately retrieve the control by sending ``` obtain control ``` command. That means I will not able to manual override for ever in position-based mode, which I observed in the first post.

While this kind of program might be naive and unexpected for OSDK's developer, I think the OSDK should consider all kinds of programs. Whatever kind of program is implemented on onboard PC, there should be a absolute way to override by RC for safety.

rohitsantdji, 2017-03-21:
@sf624 : Apologize for the delay, we were running some tests internally to figure out if we could provide a workaround to your issue.

We apologize that we are our design does not cover your particular use case, and more so that we have not documented this adequately in order to allow you to anticipate this issue and work around it. We will update the docs soon and strive to not repeat this in the future.

Regarding your use case - could you tell us more about why you want to programmatically take control whenever the drone is in a mode that allows SDK control? We would like to understand this further so that we may take it into consideration for future designs, or suggest an alternative way we might be able to help you.

closed 2017-03-30 by amenonDJI