dji-sdk/Onboard-SDK-ROS #86: Send data from Onboard-SDK to Mobile-SDK via Data Transparent Transmission

lovesmall, 2017-05-04:
Hi, I just want to know how can we send data from OSDK to MSDK in ROS.

I saw in #64 , @amenonDJI said it's not supported then. Is it possible now ? 

Looking at the demo code, seems like [ROSAdapter](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3109a6da8c33be25b91b1165721f1997ab771090/dji_sdk/include/dji_sdk/DJI_LIB_ROS_Adapter.h) is capable of sending to mobile:

```
void sendToMobile(uint8_t *data, uint8_t len){
    coreAPI->sendToMobile(data, len, NULL, NULL);
}
```

But I'm not able to using it, as not sure how to init a ROSAdapter in the demo program correctly. 

Appreciate if anyone could enlighten me.

hanzheteng, 2017-05-05:
The same problem. We find that the parameter types are string and int. And we successfully init a ROSAdapter with the serialport name of manifold and a default baudrate, and catkin_make works well. But we still could not get message on mobile device.  
Documentation -> ROS -> (serial_name: /dev/ttyTHS1 (default on Manifold) baud_rate: 230400)

hanzheteng, 2017-05-07:
We get the message today. There was something wrong with the 'if' condition last time. 
Four steps to send a message to mobile:
1. declare the pointer `rosAdapter`
2. init it with serial_name and baud_rate
3. init `data` and `len`
4. `rosAdapter->sendToMobile(&data, len)`

hanzheteng, 2017-05-16:
Finally, we find that there is another way seems like better than this. 

If you want to send a message to mobile, you can ask for the service supplied by `dji_sdk`. 

`SendMsg = nh.serviceClient<dji_sdk::SendDataToRemoteDevice>("dji_sdk/send_data_to_remote_device");`

If you want to get messages from mobile, you just need to subscribe the topic 

`MobileMsg = nh.subscribe("/dji_sdk/data_received_from_remote_device", 1, &GetMobileMsgCallback);`

lovesmall, 2017-05-18:
Hi @hanzheteng ,

Thanks for the guide, we have succeed with the send_data_to_remote_device. 

Appreciate a lot!

closed 2017-05-18 by lovesmall

bthomp2000, 2018-02-09:
Hi, 
I have tried sending data (just the number 10 here) with the following code with no success: 
```
string serial_name = "/dev/ttyTHS1";
unsigned int baudrate = 230400;
ROSAdapter* rosAdapter = new ROSAdapter();
rosAdapter->init(serial_name,baudrate);
   
uint8_t data = 10;
rosAdapter->sendToMobile(&data,1);
```

Am I missing something? And for the other example: 

`SendMsg = nh.serviceClient<dji_sdk::SendDataToRemoteDevice("dji_sdk/send_data_to_remote_device");`

How do we use this to send a message? Do we have to take further action on the returned variable SendMsg? Thanks!

tngan, 2018-03-10:
@bthomp2000 I am also working on the Data Transparent Transmission as well, it seems like mobile device can obtain the message from OES through the function `didReceiveDataFromExternalDevice`.

hanzheteng, 2018-03-19:
@bthomp2000 Do not try to use rosAdapter and those low-level codes to send message. The better way is to use ROS resources provided by DJI SDK. The reason why we use ROS is that it could provide a universal and easy way to program on robots. Please try to learn a bit more and understand how ROS work. The best resource to learn is the [ROS Wiki](http://wiki.ros.org/), or you may also find some [tutorial book like this](https://www.cse.sc.edu/~jokane/agitr/agitr-letter.pdf).

@tngan @bthomp2000  You may refer to some example programs we did before. We have shared all our source code of both [onboard ROS program](https://github.com/CruiserProject/Cruiser-OnboardROS) and [mobile program](https://github.com/CruiserProject/Cruiser-MobileController) on this [github organization](https://github.com/CruiserProject). Notice that the SDK version we used is a bit old.