dji-sdk/Onboard-SDK-ROS #87: Change Max Speed in Waypoint Control

suarez-ramon, 2017-05-17:
When sending gps waypoint lists the velocity from one point to the other is very big which means abrupt movement of the drone. Is there a config file or a way to change the max speed to make this movement less abrupt?

Thanks!

amenonDJI, 2017-06-28:
@suarez-ramon Which version of Onboard SDK, aircraft and platform are you using? 

The Waypoint Initialization allows for a "Max Velocity" parameter before you upload any of the GPS points. If you're using Onboard SDK 3.3 - take a look at the linux samples - mission_sample.cpp 
``
void
setWaypointInitDefaults(WayPointInitSettings* fdata)
``
This function initializes the max velocity parameter you're interested in.

amenonDJI, 2017-08-23:
@suarez-ramon Closed due to inactivity. Please re-open if you continue to face any issues.

closed 2017-08-23 by amenonDJI