dji-sdk/Onboard-SDK-ROS #91: SDK hangs on activation.

ilia-sheremet, 2017-05-25:
Autopilot: A3.
PC: Odroid U4, Linux  3.10.96 #1 SMP PREEMPT Thu Feb 25 13:43:04 BRT 2016 armv7l armv7l armv7l GNU/Linux

20-30% of the time activation fails and the node hangs at this line:
[`serialDevice->lockMemory();`](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk_lib/src/DJI_Link.cpp#L504)
Looks like a mutex deadlock.

When I put some debug information before (several printfs inside these functions - [CoreAPI::sendInterface()](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk_lib/src/DJI_Link.cpp#L422), [void CoreAPI::send()](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.2/dji_sdk_lib/src/DJI_API.cpp#L121)) the chance for successful activation increase.
My guess that it creates extra time to execute commands and influence deadlock timings.

ilia-sheremet, 2017-05-31:
Hello? Are you aware of this issue?

amenonDJI, 2017-06-28:
@ilia-sheremet Unfortunately the Odroid is not one of our supported platforms and we have not run any tests to confirm the OSDK working on the Odroid. We only support X86 based platforms, NVidia Jetson TK1, TX1, TX2 and the STM32.

closed 2017-06-28 by amenonDJI