dji-sdk/Onboard-SDK-ROS #92: Onboard computer

rocklinsuv, 2017-05-31:
Hi there,

I'm planning to buy an M600 for research purpose since I need more payload capacities. I would like to know if an Onboard Embedded System is needed for using the SDK? If so, any recommendation? Will an Odroid computer running ROS do the job? Many thanks!

Cheers,

ilia-sheremet, 2017-06-01:
DJI SDK is pretty lightweight. Takes around 20% of 1 CPU power on Odroid XU4 (where there are 8 CPUs). So it can work basically on everything that runs Linux.

rocklinsuv, 2017-06-01:
Thank you for the quick reply. I assume that the DJI-SDK is pretty much like the Asctec_MAV_Framework stack (or package) which runs on the Asctec onboard computers, offering a variety of functionalities. Correct me if I'm wrong :D. Thanks again.

ilia-sheremet, 2017-06-01:
Never worked with Asctec_MAV_Framework, but on the first glance it look similar.
You're welcome!

closed 2017-06-01 by rohitsantdji