dji-sdk/Onboard-SDK-ROS #93: Manifold Always saying Obtain control running after Mobile_SDK mission

Cuijie12358, 2017-06-02:
Hi guys, We found that when we use "onExecutionFinish()" on mobile_SDK to end our mobile mission and tell manifold to take control of M100,it would always saying "Obtain control running" unless we switch RC to P and back to F mode.
Does anyone knows what to do with it ?Manifold uses "request_sdk_permission_control()" to ask for  permission.

amenonDJI, 2017-06-28:
@Cuijie12358 Did you continue to see "Obtain Control Running" after re-sending Obtain Control command from Manifold? 

We've tested this on our A3/N3 platform and the transition from Mobile to Onboard is quite seamless. A quick test would be to run the Mobile Sample application and test to see if Onboard SDK control can be re-obtained after you end the Mission or after you end the Virtual Stick mode.

hanzheteng, 2017-07-03:
Hi @amenonDJI . I am the teammate of @Cuijie12358 . We did continue to see "Obtain Control Running" on the screen, with a frequency approximately over 5 Hz. 

We make a protocol between manifold and mobile for our own project. Mobile App sends a protocol code to Onboard, and a ROS node could decode it and execute the corresponding mission. 

We want to implement a function that when Waypoint Mission ended, M100 could land via X3 camera by the algorithm we designed. So we send a protocol code to manifold in function `onExecutionFinish()`, so that it could obtain control and execute this mission. 

During the test we found that manifold could not get control after Waypoint Mission, with "Obtain Control Running" on the screen. We also tried delay a few seconds and then send the command, but it did all the same. We also set a button trigger on mobile app to send the landing protocol code only for testing. The result is that, without Waypoint Mission, the landing program works well. Once we execute Waypoint Mission for only once time, then manifold could not get control. 

It seems like Mobile has a higher priority than Manifold and did not release the control after Waypoint mission. Because if we switch F mode to P and switch back immediately, then touch the trigger button, the landing program still works well.

hanzheteng, 2017-08-19:
@amenonDJI , How is it going now?

amenonDJI, 2017-08-23:
@hanzheteng Apologize for not responding earlier. I am investigating this issue and will get back to you soon.

amenonDJI, 2017-08-24:
@hanzheteng I was able to reproduce the issue. Could you run the Mobile API to **disable virtual stick** once you finish the Waypoint Mission? The Mobile SDK will then release control of the aircraft and the Onboard SDK will then be able to take control. 
``
- (void)setVirtualStickModeEnabled:(BOOL)enabled withCompletion:(DJICompletionBlock)completion
``

Cuijie12358, 2017-08-27:
@amenonDJI Thanks a lot! But we found that sometimes onExecutionFinish() or. onExecutionStart() was not called.It happens randomly when we start waypoint mission.

hanzheteng, 2017-08-28:
Released control successfully! Thanks!

[image]

Cuijie12358, 2017-08-29:
@amenonDJI I 'm sorry to disturb you, but do you know anything about this?

paperjack93, 2019-01-15:
This is happening on the official DJI GO app.

closed 2021-06-25 by dji-kyle