dji-sdk/Onboard-SDK-ROS #95: msg file missing

ChrisYin, 2017-06-16:
WaypointList.msg and MissionHotpointTask.msg are missing in dji/msg directory.
```
add_message_files(
  FILES
  Gimbal.msg
  Waypoint.msg
  ##WaypointList.msg
  MobileData.msg
  MissionWaypointAction.msg
  MissionWaypoint.msg
  MissionWaypointTask.msg
  ##MissionHotpointTask.msg
)
```
It can be find in the old version of Onboard-SDK-ROS.

groundmelon, 2017-06-16:
I meet the same problem.
https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.3/dji_sdk/srv/MissionHpUpload.srv#L1
Delete catkin_ws/build and catkin_ws/devel can reproduce this problem.
Already generated headers from old version will remain in the include directories although you delete the message file in the new version

amenonDJI, 2017-06-17:
@groundmelon @ChrisYin Sorry about the missing messages. The issue has been fixed. Please pull to get the latest. Thanks.

closed 2017-06-17 by amenonDJI