dji-sdk/Onboard-SDK-ROS #98: [Help] Error activate, Drone version not obtained in sdk 3.3 ROS

nilzamora, 2017-06-20:
When i I try to launch the sdk.launch this error appears:

STATUS/1 @ init, L45: ...Serial started successfully.
[ERROR] [1497959030.314131388]: Failed to set up port for timed read.
This usually means the serial port is not correctly set up; 
however on a small number of machines this error can come up
even when the port is correctly set up. If you are absolutely certain
your connections are okay, try commenting out L77-L89 in file dji_sdk_node.cpp
and build again.

I am commenting out those lines because my port is correctly setted up and the I get this other error:

STATUS/1 @ init, L45: ...Serial started successfully.
ERROR/1 @ parseDroneVersionInfo, L368: Drone version was not obtained. Please restart the program or call getDroneVersion

ERROR/1 @ getDroneVersion, L913: Drone version not obtained! Please do not proceed. 
Check if your serial connection is okay, and if your activation status is okay.

ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
[ERROR] [1497959254.224701157]: Vehicle initialization failed

I tried everything, so I'm sure that there's no problems of set up... 
Can anyone tell me if there's an error with the code or it's a hardware problem.
Thank you

lukvacek, 2017-06-20:
We too have experienced this issue using an N3 autopilot, the latest DJI SDK and Ubuntu 16.04.

rohitsantdji, 2017-06-20:
@nilzamora 

> I am commenting out those lines because my port is correctly setted up and the I get this other error:
If commenting out those lines doesn't immediately get the OSDK working, it usually means the error is a real communication issue. This behavior, and the subsequent messages, implies that your port has in fact not been set up properly to communicate with the OSDK. Could you provide some more information on the following topics to help continue the debug process?

1. What OSDK computer is being used, and what OS is it?
2. Do you have permissions to write to the serial port (did you run `sudo usermod -a -G $USER` ?)
3. Does OSDK 3.2 correctly start communicating with your system? (The functionality won't work correctly with the newest firmware, but the communication should.)
4. After upgrading firmware on your A3/N3, did you ensure that the baud rate to the correct value, and reboot the flight controller after clicking enable API control?
5. What baud rate are you trying to communicate at?

@lukvacek If you had the exact same issue, could you provide this info as well? If you experienced something different along the way let's hear about that too.

Thanks!

JonathanHoogvliet, 2017-06-21:
@nilzamora I also have the same exact issue. I receive the same screen error output. The same procedure works fine for the previous OSDK (3.2) (which correctly activates with the sdk_manifold.launch). After running the node itself, with baud_rate 230400, I receive the following output:

$ rosrun dji_sdk dji_sdk_node _baud_rate:=230400
>>
STATUS/1 @ init, L35: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS/1 @ init, L45: ...Serial started successfully.
STATUS/1 @ parseDroneVersionInfo, L525: Hardware = 
STATUS/1 @ parseDroneVersionInfo, L526: Firmware = 3.1.100.0
STATUS/1 @ parseDroneVersionInfo, L529: Version CRC = 0x0
ERROR/1 @ initVersion, L317: 
ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
[ERROR] [1498049049.639143969]: Vehicle initialization failed
>>

Now, I have not upgraded the A3 flight controller, and it is currently running firmware version 3.1.100.0. Baud rate, app_id, and app_key are all set correctly in the sdk.launch.

@rohitsantdji Are you implying that a firmware update is mandatory before being able to run OSDK 3.3? And if I do so, will the old OSDK (3.2) also still work with this newer firmware?

rohitsantdji, 2017-06-21:
@JonathanHoogvliet Yes, you need to upgrade (This is documented on the website [here](http://developer.dji.com/onboard-sdk/documentation/appendix/versioning.html)). Old OSDK 3.2 will not work (correctly) with newer firmware, and OSDK 3.3 will not work at all with older firmware. 

The SDK seems to be communicating correctly with the drone and receiving firmware info, so an upgrade should fix it.

c-m-jagielski, 2017-06-21:
Hello! @rohitsantdji thanks for helping. 

Here is my general information:
- NOTE I am not using ROS, I'm using the regular Linux SDK (following sample application instructions here: http://developer.dji.com/onboard-sdk/documentation/quick-start/quick-start.html) 
- Platform: Matrice 600 Pro
- Firmware package version:  1.0.1.20
- Flight controller version:  **UNKNOWN** (_How do I find this_?)
- I do not have a config.json file... I'm using UserConfig.txt since that's what's in the SDK.

Here are the answers to your questions:
#1: OS is custom RHEL6 ... a few things were upgraded to make the SDK 3.3 build/compile with cmake and make :) 
#2: Yes I have permissions and I ran the usermod command
#3: I just started my project so I downloaded 3.2.2 but never used it. Mainly I read through the code to get familiar with it. Then DJI Assistant 2 said my Matrice 600 Pro had a firmware update, which I did, and I realized the DJI Developer website was different with the 3.3 documentation so I'm using that.
#4: Yes I've rebooted everything multiple times. I've checked "Enable API Control". The baudrate on DJI Assistant 2 matches what's in my UserConfig.txt file.
#5: I'm using default baudrate of 230400

Here is the command I'm running and the program's output:
`./djiosdk-flightcontrol-sample UserConfig.txt`
  
```
Read App ID
User Configuration read successfully.
STATUS/1 @ init, L35: Attempting to open device /dev/ttyUSB0 with baudrate 230400...
STATUS/1 @ init L45: ...Serial started successfully.
ERROR/1 @ parseDroneVersionInfo, L368: Drone version was not obtained. Please restart the program or call getDroneVersion
ERROR/1 @ getDroneVersion, L913: Drone version not obtained! Please do not proceed.
Check if your serial connection is okay, and if your activation status is okay.

ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
Vehicle not initialized, exiting.
```

Thanks for your help! 
- Chris

rohitsantdji, 2017-06-21:
@c-m-jagielski - You can get the flight controller version <--> firmware package map [here](http://developer.dji.com/onboard-sdk/documentation/appendix/versioning.html). The M600 Pro firmware you are using is not compatible with OSDK 3.3 - new firmware is coming soon for the M600/Pro with support for OSDK 3.3.

c-m-jagielski, 2017-06-21:
@rohitsantdji Thanks for the info.  So when I used DJI Assistant 2, it automatically told me to get new firmware.  I'm going to try and use the 1.0.1.20 with OSDK 3.2.2.

How do I figure out my Flight Controller Version to verify it is correct (3.2.15.62)?

c-m-jagielski, 2017-06-22:
@rohitsantdji Also, can you please accurately reflect that 3.3 does not work with the M600 on the official 3.3 release notes? (http://developer.dji.com/onboard-sdk/documentation/appendix/releaseNotes.html) 

That way developers like me won't lose days working with the wrong software :)

rohitsantdji, 2017-06-26:
@c-m-jagielski Added some doc notes, as well as a version table on this github readme, mentioning the current lack of M600 support. 

For confirming your FC version, you can look at the output of the version string when you run an OSDK app - though the FW package version <--> FC version is one-to-one (i.e. if you have the 1.0.1.20 package, you are guaranteed to have the 3.2.15.62 FC firmware).

Damowerko, 2017-06-26:
Did anyone here get 3.3 working on the N3?

rohitsantdji, 2017-06-26:
@Damowerko our in-house N3s work as expected with OSDK 3.3. Could you provide the debug information requested earlier in this thread?

drieshulens, 2017-07-04:
Same problem with N3 (latest firmware), Onboard-SDK-ROS (3.3) and raspberry Pi (works fine with matrice M100 and older SDK).

This is my output:

started roslaunch server http://raspberrypi:46677/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: XXXXX <- ID
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: XXXX -< key
 * /dji_sdk/serial_name: /dev/ttyACM0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [1417]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 471e7cc0-6096-11e7-9812-b827eb9fe30c
process[rosout-1]: started with pid [1430]
started core service [/rosout]
process[dji_sdk-2]: started with pid [1433]

STATUS/1 @ init, L37: Attempting to open device /dev/ttyACM0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.
[ INFO] [1499158423.236668811]: Succeeded to read from serial device

STATUS/1 @ init, L37: Attempting to open device /dev/ttyACM0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.

ERROR/1 @ parseDroneVersionInfo, L368: Drone version was not obtained. Please restart the program or call getDroneVersion

ERROR/1 @ getDroneVersion, L913: Drone version not obtained! Please do not proceed. 
Check if your serial connection is okay, and if your activation status is okay.

ERROR/1 @ functionalSetUp, L110: Failed to initialize Version! Please exit.
[ERROR] [1499158428.238443290]: Vehicle initialization failed

Kind regards

drieshulens, 2017-07-04:
Ok I am a little further.

First I used the USB connection on the N3 instead of the API connection.

Then I tried connection at 38400 baud (set in N3 and in launch file) -> no connection
Then I tried connection at 230400 baud (set in N3 and in launch file) -> now I have a connection but next error:  

```
started roslaunch server http://raspberrypi:35927/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: xxxxxxxx
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: xxxxxxx
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [2017]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 29736c84-60b5-11e7-8fcd-b827eb9fe30c
process[rosout-1]: started with pid [2030]
started core service [/rosout]
process[dji_sdk-2]: started with pid [2037]

STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.
[ INFO] [1499171687.450289632]: Succeeded to read from serial device

STATUS/1 @ init, L37: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L47: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L525: Hardware = 

STATUS/1 @ parseDroneVersionInfo, L526: Firmware = 3.2.36.8

STATUS/1 @ parseDroneVersionInfo, L529: Version CRC = 0x0

STATUS/1 @ getErrorCodeMessage, L527: 
STATUS/1 @ getCMDSetSubscribeMSG, L609: SUBSCRIBER_ILLEGAL_INPUT

STATUS/1 @ getErrorCodeMessage, L527: 
STATUS/1 @ getCMDSetSubscribeMSG, L609: SUBSCRIBER_ILLEGAL_INPUT

STATUS/1 @ getErrorCodeMessage, L527: 
STATUS/1 @ getCMDSetSubscribeMSG, L609: SUBSCRIBER_ILLEGAL_INPUT

STATUS/1 @ getErrorCodeMessage, L527: 
STATUS/1 @ getCMDSetSubscribeMSG, L609: SUBSCRIBER_ILLEGAL_INPUT

STATUS/1 @ getErrorCodeMessage, L527: 
STATUS/1 @ getCMDSetSubscribeMSG, L609: SUBSCRIBER_ILLEGAL_INPUT

STATUS/1 @ activate, L834: version 0x3022408

STATUS/1 @ getErrorCodeMessage, L527: activate
STATUS/1 @ getCMDSetActivationMSG, L575: NEW_DEVICE_ERROR

ERROR/1 @ activate, L854: Failed to activate please retry SET 0x0 ID 0x1 code 0x3
[ERROR] [1499171688.851885504]: drone activation error
[ERROR] [1499171688.852335813]: Vehicle initialization failed
```

rohitsantdji, 2017-07-05:
@drieshulens Are you using the API connection now? The usb connection is not compatible with OSDK. Was the activation successful? Please post the output of your `roslaunch dji_sdk sdk.launch` with the API connection to help us debug this with you.

drieshulens, 2017-07-05:
@rohitsantdji My latest error with the demo is **solved**. I'm using a T14SG controller with the N3. The trust of my controller was at -10000 when i got previous error. When I put the trust stick in the middle ( at 0 ) the demo runs OK. I think this is because the DJI controller his trust is at zero by default and only -10000 when disarming ?

zliDJI, 2017-07-06:
Hi @drieshulens,  can you set your baud at 921600 in dji assistant 2 and sdk.launch? You will most probably encounter issue with 230400.

kkoppul2, 2017-07-07:
I am using a Matrice 100 with the Manifold OES and finding the same issue mentioned above. I am using SDK v3.2 and the right firmware version according to the versioning chart posted above. Firmware package 1.3.1.0, I'm not sure how to check the flight controller firmware status.  API control is enabled in the Assistant application.

ubuntu@tegra-ubuntu:~/catkin_ws/src/Onboard-SDK-ROS/dji_sdk$ roslaunch dji_sdk sdk_manifold.launch 
... logging to /home/ubuntu/.ros/log/11a15224-6349-11e7-b673-60601f199aea/roslaunch-tegra-ubuntu-2421.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tegra-ubuntu:51267/

SUMMARY
========

PARAMETERS
 * /dji_sdk/app_bundle_id: Not required in t...
 * /dji_sdk/app_id: *******
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: *******
 * /dji_sdk/groundstation_enable: 1
 * /dji_sdk/serial_name: /dev/ttyTHS1
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [2432]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 11a15224-6349-11e7-b673-60601f199aea
process[rosout-1]: started with pid [2445]
started core service [/rosout]
process[dji_sdk-2]: started with pid [2456]
=================================================
app id   : *******
app key  : *******
=================================================
--- Connection Info ---
Serial port: /dev/ttyTHS1
Baudrate: 230400
-----
STATUS init,line 62: Open serial device /dev/ttyTHS1 with baudrate 230400...
[ INFO] [1499455115.022988311]: Succeed to create thread for readPoll
=================================================
Hardware : 
Firmware : 0x00
=================================================
ERROR activate,line 449: Please call getDroneVersion first.
Broadcast call back received 

The App ID and Key are correctly read from the launch parameters.
Is this caused by a communication failure? If so how can I test the UART_CAN2 connection? I followed the Hardware guide posted in the developer guides.

adamwathieu, 2017-07-11:
@drieshulens , how did you get your configuration to work on your pi? I have a Pi3, and an M100. More info here:
https://github.com/dji-sdk/Onboard-SDK/issues/143

chutsu, 2017-07-12:
@nilzamora was wondering if you could provide an update to whether commenting the lines you mentioned in your original post affects the functionality of the SDK?

chutsu, 2017-07-12:
I have an interesting development, my setup:

- Odroid XU4 with both Onboard-SDK and Onboard-SDK-ROS installed
- Odroid <--> DJI N3, connected via USB-UART as per [docs](https://developer.dji.com/onboard-sdk/documentation/development-workflow/hardware-setup.html#a3-n3-m600-pc-linux-machine)

I'm also having problems activating but via the ROS SDK, BUT! If I execute the `djiosdk-flightcontrol-sample` from the Onboard-SDK using my own App ID and App Key I was able to execute the "Monitored Takeoff + Landing" so there might be something in the ROS SDK that prevents the code from getting activated.

chutsu, 2017-07-12:
Ok, I think I might have got it, somehow **if I plugged the USB-UART directly into the Odroid's USB port (not to a USB Hub), it works**. Was wondering if someone can give this a try and comment here.

I have a suspicion that it has something to do with UART timings via USB hubs, but I can't confirm that.

wangshuang870039239, 2017-08-16:
Have you resolved the problem? @nilzamora

cxkxc, 2018-08-23:
https://github.com/dji-sdk/Onboard-SDK-ROS/issues/98#issuecomment-313808797
Could you tell me how did you solved that problem?I get the same problem.

closed 2021-06-25 by dji-kyle