dji-sdk/Onboard-SDK-ROS #99: Where are all the messages in 3.3?

davehaddon, 2017-06-26:
There seems to be a lot of the ROS messages missing in V3.3.  
How are we supposed to get the acceleration/compass/velocity values from the aircraft now? 
Is it still possible to do attitude control with the OnboardSDK?

Damowerko, 2017-06-26:
1. Velocity: `/dji_sdk/velocity`
1. Compass: Not there as far as I can see.
1. Acceleration: `/dji_sdk/acceleration_ground_fused`
1. Attitude control is possible using `/dji_sdk/flight_control_setpoint_generic` (sensor_msgs/Joy)`.

amenonDJI, 2017-06-26:
@davehaddon In an attempt to conform with REP 103 standards, we've moved most of our custom messages to standard messages. 

Compass/Mag topic is not supported for the current release. Is there a use case for which you are interested in using compass data? 

As @Damowerko linked, all documentation can be found on the [ROS wiki](http://wiki.ros.org/dji_sdk)

Damowerko, 2017-06-26:
There are a couple more things missing like local position and tf transforms. More importantly the compliance with REP103 is limited. For example attitude commands are non-compliant. Attitude in the `base_link` frame is ambiguous. More specifically, according to REP103 rotation should be represented by:
1. Quaternion or
1. Rotation matrix or
1. Fixed axis roll, pitch, yaw about X, Y, Z or
1. Euler angle yaw, pitch, roll around Z, Y, X (which is discouraged).

The SDK represents rotation in either the base_link or the global frame. As far as I understand using `base_link` would mean that the current attitude is always (1,0,0,0). On the other hand, rotation the the global frame ("ground_ENU") implies pitch, roll, yaw euler angle rotation, which is not REP103 compliant. 

At least, this is how I understand this from the wiki.

EDIT: Thinking about it now, ground_ENU could just be the origin and the rotation is actually roll, pitch, yaw?

zliDJI, 2017-06-26:
Hi, Damowerko. Thank you for pointing the OP to the ros wiki for documentation.

Regarding local position, it's a convenient feature to have and we will add it soon. This is not related to REP103 (frames and units). Note that /dji_sdk/gps_position is the fused global position (not the gps raw data), so one can easily transform this gps_position to local position using a proper origin and conversion algorithm based on their needs.

Regarding you comment on rotation: we publish the rotation in both /dji_sdk/attitude (/geometry_msgs/QuaternionStamped, 100Hz) and inside /dji_sdk/imu (/sensor_msgs/Imu, 400Hz). In both cases, the rotation is the quaternion representation of the rotation matrix of the FLU body frame relative to the ENU world frame. 

This is the same as how popular packages such as mavros publishes attitude, such as line 208-213 in
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/imu_pub.cpp

Damowerko, 2017-06-26:
@zliDJI Thank you for clarifying gps_position, so this is the equivalent of /mavros/global_position/global.

I was referring to attitude control (subscriber rather than publisher), which is in euler angles using sensor_msgs/Joy (which adds to the confusion, since mavros uses PoseStamped, TwistStamped etc. for setpoints). Probably should have specified.

Also, where is the origin of the ENU "ground" (map?) frame? Typically that is the starting position, but for some uses it becomes the UTM zone origin.

closed 2017-07-10 by zliDJI

rahuldapp, 2018-08-07:
In reply to @amenonDJI, I am interested in using the compass data for my application. I don't see a way how so far. Can anyone help?

MikeK4y, 2018-08-31:
You can find a version of the latest SDK on my profile where the magnetic field data is also published. It's set to work with the Matrice 100 but it should be easy to adapt it to another platform