dji-sdk/Onboard-SDK-ROS #102: /dji_sdk/rc no data with sbus

acdriec, 2017-07-05:
I use a remote controller with SBUS. I can control te drone in Assistant 2 simulation an have the ROS SDK running. I can read every message, but in /dji_sdk/rc is not the data from my remote controller ? The values in there stay fixed.

```
header: 
  seq: 26307
  stamp: 
    secs: 1499263253
    nsecs: 896992701
  frame_id: rc
axes: [0.004399999976158142, -0.00139999995008111, 0.0, -1.0, 10000.0, 0.0]
buttons: []
```

zliDJI, 2017-07-06:
Hi, @acdriec 
The values [0.004399999976158142, -0.00139999995008111, 0.0, -1.0, 10000.0, 0.0] is your RC reading in [roll pitch yaw throttle, mode, gear] channel, with roll, pitch, yaw, throttle normalized between -1 and 1 (divided by 10000).  

You mean the value are not changing as you move the sticks? Could you check you connections? BTW what is your hardware setup, like flight controller?
Thanks!

acdriec, 2017-07-06:
Hello @zliDJI ,
I'm using the N3 as flight controller. when I move the sticks the values in /dji_sdk/rc are not changing, although the UAV is flying correct in simulation, so the connections are ok. 

Thanks for your help.

zliDJI, 2017-07-06:
Can you run rostopic echo /dji_sdk/rc, or rostopic hz /dji_sdk/rc to verify that the message is being updated with fixed value at around 50 Hz?  Also, in DJI Assistant 2, SDK tab, did you set the frequency of remote controller channel data to 50Hz?

acdriec, 2017-07-06:
Ok, in DJI Assistant frequency of remote controller channel data = 50Hz. With rostopic hz /dji_sdk/rc I get an average rate of 23Hz. Note that the Onboard-SDK-ROS is running on a raspberry pi 3 so that can explain the slower rate.

zliDJI, 2017-07-06:
Hi, could you set your baud rate to 921600 in both dji assistant 2 and in sdk.launch file? I just tested with the same T14SG and could read /dji_sdk/rc without any problem.

acdriec, 2017-07-06:
Hi,

When I change my baud rate to 921600 I can read /dji_sdk/rc at 50Hz so that is good! But isnt it strange that when the baud rate is 230400 the data is not changing in /dji_sdk/rc ? Only when I change the remote controller channel data frequency to 100Hz the data changes correctly.

zliDJI, 2017-07-06:
Since we are sending /dji_sdk/imu at 400 hz and some other stuff at 100hz, our channel gets saturated so you cannot get data correctly.  However, the combination 921600(baud) + 50Hz(rc) should work. You should not need to change rc to 100Hz.

acdriec, 2017-07-06:
Ok thank you for helping !

closed 2017-07-10 by zliDJI